volumio/Volumio2 #936: Various problems

trebsu, 2016-12-28:
The Album shows in compilations all different Artists as different Albums. The TAG "Albumartists" is read incorrectly.

Another mistake

If the data source, here the NAS fails and one chooses a title in volumio, Volumio crashes. Actually such inputs should be intercepted so the system does not crash.
Although startmelody on "off" it is played.

If volumio is then either restarted, or can be operated again after several minutes, it takes an eternity until internetradio activated, the switch of the radiosender is then also very sluggish.

Also, the network path is then no longer found. Only when re-storing is the source completely read again.

2lepus, 2017-01-02:
See also Issue #806 or #814

gkkpch, 2017-01-02:
It is good practise to mention at least which board you are using (PI, Odroid, X86) and which version of the Volumio image.
I take "Another mistake" as "I found more issues" :wink: 
- failing NAS, needs to be investigated, agree, should not crash Volumio
- start melody, this has been reported before. Will be fixed but has lower priority.  
- internet radio after restart, please define "eternity" and "sluggish" a little more precise.

closed 2017-02-22 by volumio

xipmix, 2017-02-24:
I can confirm the NAS issue. To reproduce (2.041):
  * connect a share in MyMusic, let the scan complete
  * ssh in and umount the share
  *  browse to the music library and try to play a track from that share
```
volumio[1405]: info: CoreCommandRouter::volumioAddQueueItems
volumio[1405]: info: [1487945830790] CoreStateMachine::addQueueItems
volumio[1405]: info: [1487945830798] CorePlayQueue::addQueueItems
volumio[1405]: info:  uri=music-library/NAS/testrm2/Rattle_and_Hum/03-Desire.flac, title=Desire, service=mpd
volumio[1405]: info: First index is 3
volumio[1405]: info: ADDING THIS ITEM TO QUEUE: {"uri":"music-library/NAS/testrm2/Rattle_and_Hum/03-Desire.flac","title":"Desire","service":"mpd"}
volumio[1405]: info: Exploding uri music-library/NAS/testrm2/Rattle_and_Hum/03-Desire.flac in service mpd
volumio[1405]: info: ----------------------------/mnt/NAS/testrm2/Rattle_and_Hum/03-Desire.flac
volumio[1405]: fs.js:987
volumio[1405]: return binding.stat(pathModule._makeLong(path));
volumio[1405]: ^
volumio[1405]: Error: ENOENT: no such file or directory, stat '/mnt/NAS/testrm2/Rattle_and_Hum/03-Desire.flac'
volumio[1405]: at Error (native)
volumio[1405]: at Object.fs.statSync (fs.js:987:18)
volumio[1405]: at Object.statSync (/volumio/node_modules/graceful-fs/polyfills.js:289:22)
volumio[1405]: at ControllerMpd.scanFolder (/volumio/app/plugins/music_service/mpd/index.js:2006:25)
volumio[1405]: at ControllerMpd.explodeUri (/volumio/app/plugins/music_service/mpd/index.js:1932:27)
volumio[1405]: at CoreCommandRouter.explodeUriFromService (/volumio/app/index.js:1098:22)
volumio[1405]: at CorePlayQueue.addQueueItems (/volumio/app/playqueue.js:140:52)
volumio[1405]: at CoreStateMachine.addQueueItems (/volumio/app/statemachine.js:150:24)
volumio[1405]: at CoreCommandRouter.addQueueItems (/volumio/app/index.js:383:27)
volumio[1405]: at Socket.<anonymous> (/volumio/app/plugins/user_interface/websocket/index.js:131:36)
mpd[613]: client: [2] closed
```
I'm guessing this needs an exception handler?
```
// app/plugins/music_service/mpd/index.js
...
ControllerMpd.prototype.scanFolder=function(uri)                                
{                                                                               
    var self=this;                                                              
    var uris=[];                                                                

+  try {                                                                            
    var stat=libFsExtra.statSync(uri);                                          
+  } catch(err) {
+    defer.resolve({}); // more thought needed here
+  }                                                                               
```
I have been thinking about this issue of the music library being out of date with respect to what shares are mounted, which has come up a few times (eg #296). Something that might be useful would be to style the track titles in a different color or strikethrough font if they fail to play or if