volumio/Volumio2 #939: Feature request: media key support

martin2250, 2017-01-02:
It would be really cool if there was an option for Volumio to react to button presses (HID keyboard buttons).
I'd just use triggerhappy together with the 'volumio' command, but there are some problems:
-the 'volumio' command takes almost a full second to connect and execute the command
-updating apt-get and installing triggerhappy led to 'No space left on device' errors
-I couldn't find a way to change volume incrementally from the command line

this is also the reason why I'm working on mpdemulation, which would be fast enough, but I'm also running into problems there (can't find a way to do incremental volume, can't install triggerhappy, 'mpc toggle' does not work without a proper status report)

gkkpch, 2017-01-02:
> -updating apt-get and installing triggerhappy led to 'No space left on device' errors (on an Odroid-C1 if that matters)

there is a problem indeed, but not just the C1. I'll investigate this

gkkpch, 2017-01-02:
Can you supply more information? I just installed triggerhappy on a C1, no issue.
Was there anything else you installed. Or did you add music to the internal storage?
Could you do a "df" and a "du /data"?

gkkpch, 2017-01-02:
As for the other questions, pls. check /etc/lirc/lircrc and see if you can get any ideas out of what I did for the remote control. 
Btw. the remote is the simple Odroid one, of course you can use another one. Just needs a bit of work...

martin2250, 2017-01-02:
Hi,
regarding the space issue:
I created a new partition on the remaining space of my sd card and filled it with music. I didn't install anything besides triggerhappy.
   `volumio@volumio:~$ df -h
    Filesystem      Size  Used Avail Use% Mounted on
    overlayfs       161M  153M     0 100% /
    /dev/mmcblk0p2  1.3G  709M  522M  58% /imgpart
    /dev/loop0      354M  354M     0 100% /static
    devtmpfs        396M     0  396M   0% /dev
    tmpfs           403M     0  403M   0% /dev/shm
    tmpfs           403M  9.7M  393M   3% /run
    tmpfs           5.0M  4.0K  5.0M   1% /run/lock
    tmpfs           403M     0  403M   0% /sys/fs/cgroup
    tmpfs           403M  4.0K  403M   1% /tmp
    tmpfs           403M     0  403M   0% /var/spool/cups
    tmpfs            20M  8.0K   20M   1% /var/log
    tmpfs           403M     0  403M   0% /var/spool/cups/tmp
    /dev/mmcblk0p1   60M  9.7M   51M  17% /boot
    /dev/mmcblk0p4   58G   23G   36G  40% /media/MUSIC
    tmpfs            81M     0   81M   0% /run/user/1000`

`du /data
0	/data/INTERNAL
1.0K	/data/albumart/folder
2.0K	/data/albumart/web/Billy%20Talent/Dead%20Silence
3.0K	/data/albumart/web/Billy%20Talent
4.0K	/data/albumart/web
6.0K	/data/albumart
5.2M	/data/backgrounds
1.0K	/data/favourites
2.0K	/data/configuration/audio_interface/alsa_controller
3.0K	/data/configuration/audio_interface
2.0K	/data/configuration/music_service/mpd
2.0K	/data/configuration/music_service/webradio
5.0K	/data/configuration/music_service
2.0K	/data/configuration/miscellanea/albumart
2.0K	/data/configuration/miscellanea/appearance
2.0K	/data/configuration/miscellanea/alarm-clock
7.0K	/data/configuration/miscellanea
2.0K	/data/configuration/system_controller/volumio_command_line_client
1.0K	/data/configuration/system_controller/volumiodiscovery
2.0K	/data/configuration/system_controller/networkfs
2.0K	/data/configuration/system_controller/network
2.0K	/data/configuration/system_controller/i2s_dacs
2.0K	/data/configuration/system_controller/system
12K	/data/configuration/system_controller
41K	/data/configuration
1.0K	/data/plugins
1.0K	/data/playlist
5.3M	/data
5.3M	total`

So I reflahed the two relevant partitions and tried it again, this time I didn't get any errors about running out of space, here is df now:
`volumio@volumio:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
overlayfs       161M  140M  8.4M  95% /
/dev/mmcblk0p2  1.3G  709M  522M  58% /imgpart
/dev/loop0      354M  354M     0 100% /static
devtmpfs        396M     0  396M   0% /dev
tmpfs           403M     0  403M   0% /dev/shm
tmpfs           403M  5.7M  397M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           403M     0  403M   0% /sys/fs/cgroup
tmpfs           403M   16K  403M   1% /tmp
tmpfs           403M     0  403M   0% /var/spool/cups
tmpfs            20M  472K   20M   3% /var/log
tmpfs           403M     0  403M   0% /var/spool/cups/tmp
/dev/mmcblk0p1   60M  9.7M   51M  17% /boot
/dev/mmcblk0p4   58G   23G   36G  40% /media/MUSIC
tmpfs            81M     0   81M   0% /run/user/1000`

the commands from lircrc don't work correctly
'play' can't pause playback (toggle does)
'prev' only jumps to the beginning of a file
but much more importantly, volumio does not notice any inputs to the mpc server, which means that playback seems to continue on the website and the volume on the website does not change at all.

gkkpch, 2017-01-02:
Ok, have not tested this for a while, possibly this is now broken.
What really worries me is the filesystem size of overlayfs, 161M is way too low, used 140M is reasonable.
Looks like your data partition never got resized, we had this before but should be corrected.
Which Volumio version are you running, could you do an 'fdisk -l' ?

gkkpch, 2017-01-02:
ehehe, just noticed something: did you add a partition after flashing Volumio and before first boot!!!
This is your issue, you shouldn't. I realize why you did it, tbut there is another way of adding music to local storage

martin2250, 2017-01-02:
Yeah, I created the partition before I booted it for the first time

gkkpch, 2017-01-02:
Ok, Volumio works like this.
We limit the size of the image to limit download time and because we do not know what size of SD or emmc card is going to be used.
Upon first boot, the data partition (mmcblk0p3) is being re-sized to the maximum capacity of the disk.
The second partition is a readonly squashfs filesystem with rootfs. Together with 2 folders on the data partition this is made into an overlayfs filesystem. As a result, the original squashfs does never get changed, and the system will take data from the data partition for all chnages that are made.
So if you add a partition before first boot, the data partition cannot grow, hence your stuck as soon as enough changes or files are added.
What you need to do is let Volumio resize the disk. You can get to disk space very easily. From a PC look on the network for Volumio, it publishes a share "internal storage" which is actually on the resized data partition (/data/INTERNAL). You can drop your files in there. After that, the only thing you need to do is start a Re-scan or Update. The files will show up in your Music Library, folder INTERNAL.

martin2250, 2017-01-02:
Is there any way to keep music on a separate partition so I can re-flash the image without having to copy everything over again?

gkkpch, 2017-01-02:
Don't see one. I'm afraid you need to bite the bullet. And let that run over night perhaps?

martin2250, 2017-01-02:
Okay, I'll figure something out for copying the music.
I just need some help getting mpdemulation to work (at least those five commands) or some other way to respond to HID inputs quickly.
Can you quickly head over to #938 and give it a quick look?

gkkpch, 2017-01-02:
can't help you there, I'm a platform maintainer. Could help you with assembler(s), C and various other languages, but no node.js, sorry

martin2250, 2017-01-02:
Okay, sure. Thanks!

macmpi, 2019-09-25:
@martin2250 did you finally have any luck in integrated HID support with triggerhappy ?
Interested in related triggerhappy configuration file.

martin2250, 2019-09-25:
> @martin2250 did you finally have any luck in integrated HID support with triggerhappy ?
> Interested in related triggerhappy configuration file.

Not with Volumio (switched to MoodeAudio soon after creating this issue, now I'm only using a headless mpd without a web interface). If you only care about mpd, you can use my configuration file, though at least back then the volume shown in the web interface would not be updated.
```
$cat /etc/triggerhappy/triggers.d/media.conf
KEY_VOLUMEUP     1 /usr/bin/mpc volume +1
KEY_VOLUMEDOWN   1 /usr/bin/mpc volume -1
KEY_PLAYPAUSE    1 /usr/bin/mpc toggle
KEY_PREVIOUSSONG 1 /usr/bin/mpc prev
KEY_NEXTSONG     1 /usr/bin/mpc next
```

macmpi, 2019-09-25:
Thanks for the quick note!
beauty of media keys is that it also works with Bluetooth AVRCP ! (with following additions)
```
KEY_PLAYCD    1 /usr/bin/mpc toggle
KEY_PAUSECD    1 /usr/bin/mpc toggle
```
Need to test using `volumio command` instead of `mpc`, as this may update UI?

ashthespy, 2021-02-27:
Added `xbindkeys` in https://github.com/volumio/Build/commit/f59854855d5690b72383d63797ce31118a7a30fe and switched to in `triggerhappy` in https://github.com/volumio/Build/commit/859858f6eee0b4de0db7aa9e3dc878e08afe0fa3

closed 2021-02-27 by ashthespy