volumio/Volumio2 #942: playlist not being read properly - can't find logging

ericj79, 2017-01-04:
I am working on a python script that reads my iTunes library and generates the playlists from the library as playlists in Volumio. I am using Volumio 2.041. The playlist extractor script can be found here: https://github.com/ericj79/playlistExtractor. For many of my playlists it seems to be working properly. For a couple of playlists there is a problem. The playlist shows up in the list of playlists, but when I click on it, nothing happens. If I try to add it to the queue. It eventually times out. I tried looking at the server requests being made to see if I could track down where the issue is, but I didn't see anything there. Is there any logs that the issue may be written to so I could trouble shoot what is wrong with those playlists? Is there some way that I could turn on additional debugging?

xipmix, 2017-01-04:
If you login to the volumio box and run 'sudo journalctl -f' you will get all the logging that's available. That should give some clues as to what is going on. You may have to edit the module(s) involved in playing playlists and restart the machine. Keep a backup of modified files so you can revert if volumio doesn't start.

ericj79, 2017-01-09:
Unfortunately the playlists that work don't really look any different than teh paylists that don't work. Here is some output for one that doesn't work: 
```
Jan 09 01:10:37 volumio volumio[803]: info: CoreCommandRouter::executeOnPlugin: mpd , handleBrowseUri
Jan 09 01:10:37 volumio volumio[803]: info: CURURI: playlists/Classical
```
When I install the web sockets developer plugin for Firefox I can see that the requests goes out and instead of a list coming back, nothing really comes back. Eventually the web client seems to send an empty socket to the server, which then responds immediately with an empty reply. But the log is not showing any problems that would indicate it is failing to respond. Any ideas for where else I could look?

closed 2017-06-02 by volumio