volumio/Volumio2 #945: RPi3 + Hifiberry dac + pro - can't connect to wifi

logip87, 2017-01-10:
My plan is to build spotify via wifi + RPi3 + HifiBerry Dac + pro.
Some version can work only with cable (lan) - wlna0 exist error
I can't connect to router. (I don't have password)

volumio wifi is working. I use version 2.038

chsims1, 2017-01-10:
If the Volumio wifi hotspot is working (192.168.211.1), can you connect to it with a tablet/phone/laptop, and then set up the wifi from the Volumio UI?

macmpi, 2017-01-17:
There's a known RPi3 wifi & HifiBerry Dac + pro issue due to interference, particularly on some specific wifi channels: you may read [this](https://github.com/raspberrypi/linux/issues/1588) for more info.
So your problem may not be related to Volumio, and you may have to contact HifiBerry support.

logip87, 2017-03-17:
Newest software helps.
Still have router set to channel 10 - new router - archer c9

closed 2017-03-17 by logip87