volumio/Volumio2 #946: i2s_dacs/index.js always overwrites existing /boot/config.txt

protenhan, 2017-01-10:
I think it is not a good idea to fully overwrite the /boot/config.txt when switching i2s_dacs settings. 

Here an example to illustrate my problem:
I have a RPI2 with the official RPI touch screen and an Hifiberry AMP +. I mounted the touch screen in a way that needs the /boot/config.txt to contain `lcd_rotate=2` to rotate the screen content 180 degree to be right sight up. I also increased the gpu memory to 64MB for the touch screen to work smoothly. Now every time I change the i2c_dacs settings the whole /boot/config.txt gets overwritten. 

My suggestion for fixing this would be that the volumio app contains some sort of API that can be also be accessed by Plugins (e.g. the TouchScreen Plugin) to modify, add or delete existing values of the /boot/config.txt. 

Disclaimer: I'm not familiar with node.js so I have no idea how something like this would be implemented.

macmpi, 2017-01-16:
Indeed [writeI2SDAC](https://github.com/volumio/Volumio2/blob/master/app/plugins/system_controller/i2s_dacs/index.js#L437) should apply changes with sed-like search/replace rather than rewriting `config.txt` as a whole.

macmpi, 2017-03-29:
@xipmix wrote:

> Is it possible for there to be more than one dtoverlay= line in a /boot/config.txt file?

The only 2 modifying sources I find are `writeI2SDAC` and `disableI2SDAC` I think.
`writeI2SDAC` it's probably easy, as routine exactly knows what to add.
`disableI2SDAC` can be more tricky indeed, but actually it can only be a DAC listed in `dacs.json` (or std audio), so search/replace of exact line `dtoverlay=targettedDAC` is probably relevant, testing all possible ones (not that many).

Other option could be to put a sort of dedicated Audio section in `/boot/config.txt`, between `#***Volumio Audio output settings***` markers for instance, and nail-down adds & searches in there?

macmpi, 2017-05-14:
addressed in https://github.com/volumio/Volumio2/pull/1176

macmpi, 2017-05-30:
With merged fix, you may close this issue

closed 2017-05-30 by volumio