volumio/Volumio2 #948: unix user's password cannot be changed without breaking spotify upgrades

bablat, 2017-01-10:
I believe Volumio should be able to withstand changing the default username/password (volumio:volumio). Requiring a default password to remain unchanged is a very bad security practice and could easily allow an attacker to gain full control to any volumio device on any network.

When upgrading plugins (I use spotify), the process fails if the password is changed. When I change it back to the default the process always completes successfully.

Thanks for building this great software, hope we can get proper snapcast integration (for spotify and airplay as well!).

volumio, 2017-01-11:
Yes, I definetely agree, its a bad design practice.
What we can do is make the plugin installation work when pw is changed, but we need to keep the user volumio

bablat, 2017-01-11:
I believe that a straightforward temporary fix could be to allow running the plugin installer script under sudo with the NOPASSWD directive (e.g. http://unix.stackexchange.com/questions/18830/how-to-run-a-specific-program-as-root-without-a-password-prompt).
This would simply remove the prompt (leaving a risk there, easy escalation from volumio->root if that script is compromised) but far better than requiring password to remain unchanged.
Another option would be to prompt for password if the sudo command fails, and then retry it.

Not sure if the plugin installer even needs root access to run? I assume it's not critical for most plugins but you wanted to keep the possibility available for future plugins?

bablat, 2017-01-11:
I also believe that once this is addressed, volumio should prompt for a new password once installed. default passwords are bad even if a user is allowed to change them. I'd hate for volumio to get the wrong type of publicity: http://www.welivesecurity.com/2016/10/24/10-things-know-october-21-iot-ddos-attacks/

gonace, 2017-03-05:
Could the "volumio" user be interpreted as a "system" user and adding a "admin" user which will be prompted for a password during setup/configuration?