volumio/Volumio2 #959: NAS mount doesn't work

digx77, 2017-01-15:
Trying to add a BUFFALO NAS with below info
Name: NAS
address: 192.168.1.108
folder: share
username: admin
password: xxxxxxxxx

NAS is showed in the list of "My Music" section; but it is not Mounted (red cross showed)
Is it a known issue?

xipmix, 2017-01-17:
You will need to post some logs to make it possible to address this issue. ssh in and run 'sudo journalctl' (just after trying to add the NAS) to get the relevant log output.

digx77, 2017-01-17:
Thanks xipmix,
below the log but I cannot see any useful info. Maybe some additional steps are needed to enable proper output?

root@volumio:/home/volumio# sudo journalctl
-- Logs begin at Mon 2017-01-16 17:27:51 UTC, end at Tue 2017-01-17 08:21:48 UTC. --
Jan 16 17:27:51 volumio systemd-journal[182]: Runtime journal is using 4.0M (max allowed 30.0M, trying to leave 72.9M
Jan 16 17:27:51 volumio kernel: Booting Linux on physical CPU 0xf00
Jan 16 17:27:51 volumio kernel: Initializing cgroup subsys cpuset
Jan 16 17:27:51 volumio kernel: Initializing cgroup subsys cpu
Jan 16 17:27:51 volumio kernel: Initializing cgroup subsys cpuacct
Jan 16 17:27:51 volumio kernel: Linux version 4.4.9-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstoo
Jan 16 17:27:51 volumio kernel: CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
Jan 16 17:27:51 volumio kernel: CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Jan 16 17:27:51 volumio kernel: Machine model: Raspberry Pi 2 Model B Rev 1.1
Jan 16 17:27:51 volumio kernel: cma: Reserved 8 MiB at 0x3cc00000
Jan 16 17:27:51 volumio kernel: Memory policy: Data cache writealloc
Jan 16 17:27:51 volumio kernel: On node 0 totalpages: 253952
Jan 16 17:27:51 volumio kernel: free_area_init_node: node 0, pgdat 808c0d80, node_mem_map bc348000
Jan 16 17:27:51 volumio kernel:   Normal zone: 2232 pages used for memmap
Jan 16 17:27:51 volumio kernel:   Normal zone: 0 pages reserved
Jan 16 17:27:51 volumio kernel:   Normal zone: 253952 pages, LIFO batch:31
Jan 16 17:27:51 volumio kernel: [bcm2709_smp_init_cpus] enter (9520->f3003010)
Jan 16 17:27:51 volumio kernel: [bcm2709_smp_init_cpus] ncores=4
Jan 16 17:27:51 volumio kernel: PERCPU: Embedded 13 pages/cpu @bd770000 s22592 r8192 d22464 u53248
Jan 16 17:27:51 volumio kernel: pcpu-alloc: s22592 r8192 d22464 u53248 alloc=13*4096
Jan 16 17:27:51 volumio kernel: pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Jan 16 17:27:51 volumio kernel: Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 251720
Jan 16 17:27:51 volumio kernel: Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=41
Jan 16 17:27:51 volumio kernel: PID hash table entries: 4096 (order: 2, 16384 bytes)
Jan 16 17:27:51 volumio kernel: Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
Jan 16 17:27:51 volumio kernel: Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
Jan 16 17:27:51 volumio kernel: Memory: 979364K/1015808K available (6340K kernel code, 432K rwdata, 1712K rodata, 480K
Jan 16 17:27:51 volumio kernel: Virtual kernel memory layout:
                                    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                                    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
lines 1-31
-- Logs begin at Mon 2017-01-16 17:27:51 UTC, end at Tue 2017-01-17 08:21:48 UTC. --
Jan 16 17:27:51 volumio systemd-journal[182]: Runtime journal is using 4.0M (max allowed 30.0M, trying to leave 72.9M free of 482.4M availab
Jan 16 17:27:51 volumio kernel: Booting Linux on physical CPU 0xf00
Jan 16 17:27:51 volumio kernel: Initializing cgroup subsys cpuset
Jan 16 17:27:51 volumio kernel: Initializing cgroup subsys cpu
Jan 16 17:27:51 volumio kernel: Initializing cgroup subsys cpuacct
Jan 16 17:27:51 volumio kernel: Linux version 4.4.9-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g846061
Jan 16 17:27:51 volumio kernel: CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
Jan 16 17:27:51 volumio kernel: CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Jan 16 17:27:51 volumio kernel: Machine model: Raspberry Pi 2 Model B Rev 1.1
Jan 16 17:27:51 volumio kernel: cma: Reserved 8 MiB at 0x3cc00000
Jan 16 17:27:51 volumio kernel: Memory policy: Data cache writealloc
Jan 16 17:27:51 volumio kernel: On node 0 totalpages: 253952
Jan 16 17:27:51 volumio kernel: free_area_init_node: node 0, pgdat 808c0d80, node_mem_map bc348000
Jan 16 17:27:51 volumio kernel:   Normal zone: 2232 pages used for memmap
Jan 16 17:27:51 volumio kernel:   Normal zone: 0 pages reserved
Jan 16 17:27:51 volumio kernel:   Normal zone: 253952 pages, LIFO batch:31
Jan 16 17:27:51 volumio kernel: [bcm2709_smp_init_cpus] enter (9520->f3003010)
Jan 16 17:27:51 volumio kernel: [bcm2709_smp_init_cpus] ncores=4
Jan 16 17:27:51 volumio kernel: PERCPU: Embedded 13 pages/cpu @bd770000 s22592 r8192 d22464 u53248
Jan 16 17:27:51 volumio kernel: pcpu-alloc: s22592 r8192 d22464 u53248 alloc=13*4096
Jan 16 17:27:51 volumio kernel: pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Jan 16 17:27:51 volumio kernel: Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 251720
Jan 16 17:27:51 volumio kernel: Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2709.boardrev=0xa
Jan 16 17:27:51 volumio kernel: PID hash table entries: 4096 (order: 2, 16384 bytes)
Jan 16 17:27:51 volumio kernel: Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
Jan 16 17:27:51 volumio kernel: Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
Jan 16 17:27:51 volumio kernel: Memory: 979364K/1015808K available (6340K kernel code, 432K rwdata, 1712K rodata, 480K init, 764K bss, 28252
Jan 16 17:27:51 volumio kernel: Virtual kernel memory layout:
                                    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                                    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)

digx77, 2017-01-17:
Maybe I found info in /var/log/volumio.log as below:
......
2017-01-17T08:38:50.573Z - info: CoreCommandRouter::executeOnPlugin: networkfs , addShare
2017-01-17T08:38:50.575Z - info: Adding a new share
2017-01-17T08:38:50.576Z - info: No correspondence found in configuration for share NAS on IP 192.168.1.108
2017-01-17T08:38:54.613Z - info: CoreCommandRouter::getUIConfigOnPlugin
2017-01-17T08:38:54.641Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2017-01-17T08:38:54.654Z - info: CoreCommandRouter::executeOnPlugin: networkfs , listShares
......

I cannot understand why we have: "No correspondence found in configuration for share NAS on IP 192.168.1.108"

since I can access to this NAS via Windows file explorer.
Someone could kindly check this?

digx77, 2017-01-17:
Ok I found the issue:
Mainly if LAN and wifi is both configured (in my case LAN as DHCP and wifi with fixed IP); but wifi dongle is removed and LAN cable attached, the NAS is not reachable (very strange behaviour since other computer were reachable).

Setting only one configuration LANn or WiFi it works
It seems the problem is when a Fixed IP is setup but not used since DHCP is used on other physical connection

xipmix, 2017-01-18:
When you say 'other computer were reachable' do you mean: a) you could reach shares they provide or b) you could ping them (e.g. ```ping 192.168.1.77```). 

If a), were any of the shares already mounted? (the output of ```cat /proc/mounts``` would help.

Are the wifi and ethernet interfaces getting addresses in the same subnet (192.168.1.x)?

This seems like a corner case that may not be worth fixing. A comprehensive fix probably entails forcing an unmount and remount of all mounted shares if some network interfaces are brought down and others brought up. Probably doable, but who knows when.

digx77, 2017-02-04:
Hi
I found the exact case to reproduce the issue:
1) Have cable LAN connected with DHCP --> Volumio is reachable 
2) Connect wifi dongle and configure WiFi connection
3) Volumio works and is reachable with both LAN address and also WiFi address
4) issue, until the cable is not removed the NAS is not reachable from Volumio device so NAS cannot be added

The issue seems related to Network routing on Volumio when both LAN and WiFi is configured on same subnet (192.168.1.x).
It is not a critical and important issue but I think should be investigate

volumio, 2017-02-04:
Indeed, I think it might be related to samba configuration.. Any help on this is appreciated..

xipmix, 2017-02-05:
> 
> The issue seems related to Network routing on Volumio when both LAN and WiFi is configured on same subnet (192.168.1.x).
> It is not a critical and important issue but I think should be investigate

Can you post the output of 'sudo netstat -rn' and 'sudo ifconfig -a'
when the system is in state 3) and then in 4) (cable removed) ?

This will help understand how the routing is configured.

closed 2017-02-20 by volumio