volumio/Volumio2 #962: When in play queue, on shuffle, take song off list once played

shawaj, 2017-01-17:
There is an annoying issue in the play queue.

If i add songs to the queue from Spotify, and then play the queue with shuffle turned on...the songs are not removed from the queue as they are played.

This leads to songs being played many times over.

Thanks

Callenberg, 2017-01-17:
Hello,
Volumio 2 has no shuffle , it is a random play mode. If you hover over the button it even says "Random". In order to remove the tracks from the queue when played you have to press the consume button as well. That will remove each played track from the queue. The consume button is left of the trash can.

The difference between shuffle and random is that shuffle is supposed to rearrange the playing order and plays each track once (given that repeat mode is not on) where the random mode plays the tracks in random order indefinitely. 

Random + Consume modes should make Volumio do the thing you want. Unfortunately Volumio 2.041 doesn't work like that at the moment. Random + Consume means that the queue is played in the queue order an each track is removed after it has been played. (behaves like Consume mode only). Yes, it is is a bug.

 Best regards,

shawaj, 2017-01-18:
Ah OK, my bad. It uses the same logo as shuffle on other platforms so got
confused. Didn't notice the hover.

Thanks for clarification. Guessing this bug will be updated in future
versions?

closed 2017-06-02 by volumio