volumio/Volumio2 #973: wifi password appears in plain text

Zougi, 2017-01-21:
It would be great to encrypt wifi password stored in file _/data/configuration/system_controller/network/config.js_!

distler, 2017-01-22:
1. If you encrypt the password, then you need to store the encryption key somewhere (or prompt the user for it, which is not really feasible for an "appliance" that's supposed to operate headless).
2. The wifi password is *already* stored in plain text on your wifi router (for exactly the same reason). Since (if you're like most people) your router is directly connected to the internet (unlike the box you're running volumio on), that seems like a much bigger issue.
3. N.b. This is unlike the system password, where you only need to store a *hash*  of the password (and compare that to a hash of the user-supplied password). In WPA-PSK, the encryption key is derived from the wifi password, so every device on the network needs a plain-text copy of the wifi password.

Zougi, 2017-01-22:
whaou I learnt something today. I'll look deeper at the configuration of my router! I'm closing the issue. Thanks for the explanation distler.

closed 2017-01-22 by Zougi