volumio/Volumio2 #974: Hostname change incomplete (SMB) & errors

macmpi, 2017-01-22:
When changing hostname in System UI, SMB service does not get changed to new host name.
Indeed [setHostname](https://github.com/volumio/Volumio2/blob/master/app/plugins/system_controller/system/index.js#L220) does not seem to handle `/etc/samba/smb.conf`, nor restart that service, unlike avahi.

I enclose log of such name change as an example (change from "toto" to "test").
One can note error logs related to `/etc/hosts` permissions, and missing `/etc/avahi/services/volumio.service` (`/etc/avahi/services` directory is indeed empty and `volumio.service` seems to be at `/lib/systemd/system/volumio.service`)

```
Jan 22 19:10:32 toto volumio[789]: info: CALLMETHOD: system_controller system saveGeneralSettings [object Object]
Jan 22 19:10:32 toto volumio[789]: info: CoreCommandRouter::executeOnPlugin: system , saveGeneralSettings
Jan 22 19:10:32 toto volumio[789]: info: System name has changed, restarting Shairport Sync
Jan 22 19:10:32 toto volumio[789]: Discovery: Restarting stuff
Jan 22 19:10:32 toto volumio[789]: Discovery: Stopped ads, if present
Jan 22 19:10:32 toto sudo[1144]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/killall upmpdcli
Jan 22 19:10:32 toto sudo[1144]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 22 19:10:32 toto sudo[1144]: pam_unix(sudo:session): session closed for user root
Jan 22 19:10:32 toto sudo[1149]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/chmod 777 /etc/hosts
Jan 22 19:10:33 toto sudo[1149]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 22 19:10:33 toto sudo[1152]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/systemctl restart airplay
Jan 22 19:10:33 toto volumio[1063]: src/ohmetacache.cxx:123::dmcacheSaveWorker: can't get task from queue
Jan 22 19:10:33 toto mpd[963]: client: [2] closed
Jan 22 19:10:33 toto sudo[1149]: pam_unix(sudo:session): session closed for user root
Jan 22 19:10:33 toto sudo[1152]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 22 19:10:33 toto volumio[789]: info: Permissions for /etc/hosts set
Jan 22 19:10:33 toto systemd[1]: Stopping ShairportSync AirTunes receiver...
Jan 22 19:10:33 toto shairport-sync[1089]: exit...
Jan 22 19:10:33 toto systemd[1]: Starting ShairportSync AirTunes receiver...
Jan 22 19:10:33 toto systemd[1]: Started ShairportSync AirTunes receiver.
Jan 22 19:10:33 toto sudo[1152]: pam_unix(sudo:session): session closed for user root
Jan 22 19:10:33 toto shairport-sync[1176]: startup
Jan 22 19:10:33 toto volumio[789]: info: Hostname now is test
Jan 22 19:10:33 toto volumio[789]: info: [1485112233641] Shairport-Sync Started
Jan 22 19:10:33 toto volumio[789]: info: mDNS: A device disapperared from network
Jan 22 19:10:33 toto volumio[789]: info: mDNS: Device toto disapperared from network
Jan 22 19:10:35 toto sudo[1175]: volumio : unable to resolve host toto
Jan 22 19:10:35 toto sudo[1175]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/hostname test
Jan 22 19:10:35 toto sudo[1175]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 22 19:10:35 test sudo[1175]: pam_unix(sudo:session): session closed for user root
Jan 22 19:10:35 test volumio[789]: info: New hostname set
Jan 22 19:10:35 test sudo[1183]: volumio : unable to resolve host toto
Jan 22 19:10:35 test sudo[1185]: volumio : unable to resolve host toto
Jan 22 19:10:35 test sudo[1183]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/chmod 777 /etc/avahi/services/volumio.service
Jan 22 19:10:35 test sudo[1185]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/systemctl start upmpdcli.service
Jan 22 19:10:35 test sudo[1183]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 22 19:10:35 test sudo[1185]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 22 19:10:35 test sudo[1183]: pam_unix(sudo:session): session closed for user root
Jan 22 19:10:35 test systemd[1]: Starting UPnP Renderer front-end to MPD...
Jan 22 19:10:35 test systemd[1]: Started UPnP Renderer front-end to MPD.
Jan 22 19:10:35 test volumio[789]: Canot set permissions for /etc/hosts: Error: Command failed: /usr/bin/sudo /bin/chmod 777 /etc/avahi/services/volumio.service
Jan 22 19:10:35 test volumio[789]: sudo: unable to resolve host toto
Jan 22 19:10:35 test volumio[789]: /bin/chmod: cannot access '/etc/avahi/services/volumio.service': No such file or directory
Jan 22 19:10:35 test sudo[1185]: pam_unix(sudo:session): session closed for user root
Jan 22 19:10:35 test volumio[789]: info: Upmpdcli Daemon Started
Jan 22 19:10:35 test mpd[963]: client: [3] opened from ::ffff:127.0.0.1:49816
Jan 22 19:10:36 test sudo[1219]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/systemctl restart avahi-daemon.service
Jan 22 19:10:36 test sudo[1219]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 22 19:10:36 test systemd[1]: Stopping ShairportSync AirTunes receiver...
Jan 22 19:10:36 test shairport-sync[1176]: exit...
Jan 22 19:10:36 test systemd[1]: Stopping Avahi mDNS/DNS-SD Stack...
Jan 22 19:10:36 test avahi-daemon[1085]: Got SIGTERM, quitting.
Jan 22 19:10:36 test avahi-daemon[1085]: Leaving mDNS multicast group on interface wlan0.IPv4 with address 192.168.3.28.
Jan 22 19:10:36 test volumio[789]: info: mDNS Browse raised the following error Error: dns service error: unknown
Jan 22 19:10:36 test avahi-daemon[1085]: avahi-daemon 0.6.31 exiting.
Jan 22 19:10:36 test dbus[496]: [system] Activating via systemd: service name='org.freedesktop.Avahi' unit='dbus-org.freedesktop.Avahi.service'
Jan 22 19:10:36 test systemd[1]: Starting Avahi mDNS/DNS-SD Stack...
Jan 22 19:10:36 test avahi-daemon[1227]: Process 1085 died: No such process; trying to remove PID file. (/var/run/avahi-daemon//pid)
Jan 22 19:10:36 test avahi-daemon[1227]: Found user 'avahi' (UID 104) and group 'avahi' (GID 105).
Jan 22 19:10:36 test avahi-daemon[1227]: Successfully dropped root privileges.
Jan 22 19:10:36 test avahi-daemon[1227]: avahi-daemon 0.6.31 starting up.
Jan 22 19:10:37 test dbus[496]: [system] Successfully activated service 'org.freedesktop.Avahi'
Jan 22 19:10:37 test sudo[1219]: pam_unix(sudo:session): session closed for user root
Jan 22 19:10:37 test systemd[1]: Started Avahi mDNS/DNS-SD Stack.
Jan 22 19:10:37 test avahi-daemon[1227]: Successfully called chroot().
Jan 22 19:10:37 test avahi-daemon[1227]: Successfully dropped remaining capabilities.
Jan 22 19:10:37 test avahi-daemon[1227]: No service file found in /etc/avahi/services.
Jan 22 19:10:37 test avahi-daemon[1227]: Joining mDNS multicast group on interface wlan0.IPv4 with address 192.168.3.28.
Jan 22 19:10:37 test avahi-daemon[1227]: New relevant interface wlan0.IPv4 for mDNS.
Jan 22 19:10:37 test avahi-daemon[1227]: Network interface enumeration completed.
Jan 22 19:10:37 test avahi-daemon[1227]: Registering new address record for 192.168.3.28 on wlan0.IPv4.
Jan 22 19:10:37 test avahi-daemon[1227]: Registering HINFO record with values 'ARMV6L'/'LINUX'.
Jan 22 19:10:37 test systemd[1]: Starting ShairportSync AirTunes receiver...
Jan 22 19:10:37 test systemd[1]: Started ShairportSync AirTunes receiver.
Jan 22 19:10:37 test volumio[789]: info: Avahi Daemon Restarted
Jan 22 19:10:37 test shairport-sync[1232]: startup
Jan 22 19:10:37 test volumio[789]: Discovery: StartAdv! true
Jan 22 19:10:37 test volumio[789]: Discovery: Started advertising... test - true
Jan 22 19:10:38 test avahi-daemon[1227]: Server startup complete. Host name is test.local. Local service cookie is 2554415864.
Jan 22 19:10:39 test volumio[789]: Discovery: adding e970f3a0-51eb-4b1d-942c-40a0fea9e38a
Jan 22 19:10:39 test volumio[789]: info: mDNS: Found device test
Jan 22 19:10:45 test sudo[1239]: volumio : TTY=pts/0 ; PWD=/home/volumio ; USER=root ; COMMAND=/bin/journalctl
Jan 22 19:10:45 test sudo[1239]: pam_unix(sudo:session): session opened for user root by volumio(uid=0)
```

xipmix, 2017-02-21:
Some ramblings that may help are in #293

macmpi, 2017-03-29:
Manually changing `netbios name = <new_host>` within `/etc/samba/smb.conf` and restarting Pi (or just service probably) publishes the new SMB name.

Unclear if other setting `name resolve order = bcast lmhosts host wins` as mentioned [here](http://askubuntu.com/questions/243461/how-can-i-connect-to-a-samba-server-using-its-hostname-instead-of-the-ip) is really required.

xipmix, 2017-05-08:
@Warter21 What do you mean about WORKGROUP? Do you want to be able to set it in the UI? If so please open a separate issue.

xipmix, 2017-05-08:
@macmpi  I think the ```name resolve order``` is attempting to ensure the netbios name is used in preference to the DNS or /etc/hosts hostname (ie what nsswitch.conf stipulates). I think placing WINS last is an attempt to minimise conflicts with any Windows based WINS servers that may be on the local network. FWIW https://www.samba.org/samba/docs/using_samba/ch07.html says the default order is ```lmhosts hosts wins bcast```. Without actually testing anything, I think it's a reasonably robust approach for a home network.

macmpi, 2017-05-08:
Thanks @xipmix 
However I would not touch that `name resolve` part for now: I think just getting the `netbios name` right would be a nice benefit to start with.
Can't make the necessary change currently, but feel free to do so if you have some time & motivation.

xipmix, 2017-05-08:
I agree, one thing at a time. 

The precedence of ```playerName``` vs ```hostName``` is unclear to me. 
 * The UI provides a way to change ```playerName```. The default setting for that config item is derived from the hostname. 
 * It seems to me that ```playerName``` is intended to match the netbios name so it can be identified when browsing the network, windows style. 
 * But the only relation those two names should have to the hostname is that the hostname provides the initial value for the other two.

So it seems  ```setHostname``` shouldn't actually be getting called if the user changes the hamburger->System->Player Name setting. Or at least it should not be updating ```/etc/hostname```, but rather ```/etc/samba/smb.conf```(and I guess the mdns stuff). There could well be a case for another part of the UI changing the hostname (or 'System Name') but I haven't seen it yet.

volumio, 2017-11-24:
Fixed with 
https://github.com/volumio/Volumio2/commit/74b3b8467a5248559314fa75099ffde5ff7c0b4b
and
https://github.com/volumio/Volumio2/commit/477d1254dde7235844d5ee856a3235ade13b5032

closed 2017-11-24 by volumio