volumio/Volumio2 #975: Back button jumps back either 2 or 3 menus, not just one

shawaj, 2017-01-24:
Hitting the back button next to the home icon on Volumio UI actually takes you back 2 or 3 pages instead of just one.

xipmix, 2017-01-26:
To help with reproducing this, can you provide a specific example - what page you were on when you hit the back button and what page you ended up on? It would help also to know the version you are running. To really nail it down,  a copy of ```tail -f /var/log/volumio.log``` from just before pressing the back button would be useful also.

shawaj, 2017-01-26:
This is version 2.041

This happens on any Spotify page.

For example if I click Spotify -> Genres and Moods -> Rock -> Rock This and
then click back it takes me right back to the main Spotify page.

Thanks

or mute
the thread

volumio, 2017-01-26:
@skikirkwood do you think we can fix this one? 
We could add in the browser mechanism the property currentPos so back will be populated with this one.
What do you think?

skikirkwood, 2017-01-26:
Yes, we should definitely fix this.  I've been aware of it from day one and always found it annoying, just didn't know how to to fix it.  Let me know how when currentPos is in and I'll take care of it.

closed 2018-03-26 by volumio