volumio/Volumio2 #977: v2.041 / Rpi3 - Shutdown leads to reboot

Vigo0101, 2017-01-25:
Hello,
Setup is as follow:
- RPi3
- ES9018K2M DAC
- RPi 7" official Touchscreen with plugin installed
- Volumio 2.041 fresh setup.

When doing a shutdown either from the UI or from console using "shutdown -h -P now" or "halt", the system does not shutdown but reboot.
Doing a reboot makes the system to reboot normally.
So, it is no possible to shutdown the system in a safe way.
May be a kernel issue with systemd ?
Disconnecting the touchscreen and the DAC does not solve the problem.
Same system configuration is working fine with last Raspbian Jessie so it seems not to be a hardware issue.
Any advice ?
Thank you.

xipmix, 2017-02-07:
It's helpful to cross-link if you post to the forum as well: https://volumio.org/forum/rpi3-volumio-041-shutdown-leads-reboot-t5705.html. Could you link to this bug from there?

What happens if you give the command
```
sudo poweroff
```
instead of ```shutdown``` with arguments? I'm expecting it to be the same but worth checking. 

BTW, 'halt' will stop the machine but not power it off entirely, it's supposed to work that way but was wrongly implemented for several years.

Could there be a setting in the boot partition that tells the pi to boot whenever power is applied?

Vigo0101, 2017-02-07:
@xipmix 
Thank you for your help.

Link to this issue added in my original post in Volumio forum.

"sudo poweroff" leads to a reboot, too.

I don't know if there is something special in boot partition. Installation is made from original Volumio 2.041 download and a brand new SDcard (8 Gb Sandisk class 10).
Changing to another SDcard does not solve this issue.
Power supply voltage at Pi input seems to be correct. No glitch seen on oscilloscope.

xipmix, 2017-02-08:
I'm stumped, but I'll throw out a few thoughts that might help..

What current is that supply rated for? I think you need something that can supply a bit over 2 Amperes. Probably ok if raspbian works. What kernel version is raspbian (uname -a), it might be newer than what volumio has.

It might be worth investigating [wake_from_halt](http://elinux.org/RPI_safe_mode). 

[This page ](http://raspberrypi.stackexchange.com/questions/42929/how-to-safely-switch-off-the-raspberry-pi) describes the shutdown in some detail, you might want to check the LED behave as described. Does the reboot happen at once or is there a delay?

Vigo0101, 2017-02-09:
@xipmix 
Thank you for your tips.

Power supply can provide 3A @ 5V. Output voltage metered at 5.02 V

Raspbian kernel release is 4.1.19.
Volumio 2.041 kernel release is 4.4.9.

I have check impedance between pin 5 and pin 6. It is quite infinite. So, it doesn't seem to be a kind of wake from halt.

At shutdown, green led blinks around 10 times then goes off. Reboot occurs immediately after.

xipmix, 2017-02-09:
Hm.
Do you have any older volumio images you could try?

Otherwise, perhaps ask in the forum for other pi3 owners
who have the same problem. (volumio.org/forum)

--

xipmix, 2017-02-12:
Just fyi I found an rpi3 board and tried it out with 2.041. It says 'rpi 3 model B v1.2' on the circuit board, is yours the same model?

I have hdmi, analog audio and ethernet connected.

First I tried 'poweroff' and 'shutdown -h -P now'; both worked as expected. 
Tried a normal reboot, then a poweroff. Also worked fine.
Tried a poweroff through the web ui. Also worked fine.
Tried a reboot and then poweroff, both through the web ui. Also worked fine.
With all the tests above I waited 10s or so to see if the system would start on its own.
Also tried with hdmi unplugged, poweroff from an SSH sesssion worked fine.

All I can think of now is:
 - something weird with the power supply is causing it to appear to the pi like it is being unplugged and replugged. Possibly something plugged into the same power strip or wall socket. This is grasping at straws, this should be happening with raspbian too.
 - something in the systemd setup has gotten messed up and it is interpreting 'poweroff' as 'reboot'.  This is what I have in places that seem relevant
```
$ ls -l /etc/systemd/system/poweroff.target.wants/
total 0
lrwxrwxrwx 1 root root 40 Dec 12 00:20 hwclock-save.service -> /lib/systemd/system/hwclock-save.service
$ ls -l /lib/systemd/system/shutdown.target.wants/
total 0
lrwxrwxrwx 1 root root 21 Jul 17  2014 alsa-store.service -> ../alsa-store.service
```
I watched to see if the reboot behaviour of the green LED differs from the shutdown behaviour. That seems to be the case, the ten flashes do not occur. So this idea is probably wrong.

I assume you have tried reflashing and/or a different flash card...

Vigo0101, 2017-02-13:
@xipmix 
Again, thank you for your help.
I have dismantle my player to change the RPi 3 board with a new one.
The issue still persists with a brand new board.
Then, I put back the old RPi 3 board but without the DAC. 
Without the DAC, Volumio shutdown correctly.
So, it seems the DAC is the root cause.
I have check again impedance between GPIO pins 5 and 6, it is infinite.
I will now address this issue with Audiophonics customer support.

closed 2017-02-20 by volumio