volumio/Volumio2 #981: distorted sound on volumio2 not on volumio1 with xmos

rwrw01, 2017-01-29:
changed this weekend to volumio2 from volumio1. The sound is full of distortion, as if the stream is not wel streamed.

My dac is a xmos 9023 with a diyinhk board: hwparam:

a not so well googletranslate page seems to descripe a problem with  sabre dacs:
http://nw-electric.way-nifty.com/blog/2016/07/volumio2-sabreb.html

how can I best analyze the difference between volumio1/2. The only obvious difference I see is the absence of the "interprator" ?

gkkpch, 2017-01-30:
if you could also add more about your installation, we would not have to guess.
Please tell what board you are using, PI, Odroid, Cubox? Which type?

rwrw01, 2017-01-30:
Raspberry Pi B+.

I use the latest (official) image and performed a update/upgrade. in the volumio 1.55 build I do not see the HDMI/native soundcard, in the volumio 2 image I can select native audio out and the D20 (xmos).

The xmos 384khz usb-i2c is linked to the sabre 9023 like shown here:
http://www.raspyfi.com/building-my-reference-dac/

ralph

stationweb, 2017-11-19:
Same problem for me with the same combo: Pi B+ / xmos 384khz usb-i2c to sabre 9023