volumio/Volumio2 #988: shairport-sync periodically crashes

ormanya, 2017-02-02:
Streaming music from iTunes to my Volumio box occasionally causes shairport-sync to crash with the following error:
`Feb 02 22:41:27 office shairport-sync[982]: shairport-sync: player.c:247: alac_decode: Assertion `outsize == (4 * frame_size)' failed.
Feb 02 22:41:27 office systemd[1]: airplay.service: main process exited, code=killed, status=6/ABRT
Feb 02 22:41:27 office systemd[1]: Unit airplay.service entered failed state.`

This does not seem to be associated with any particular action (happens while playing in the background and often mid-track). After rebooting (or restarting shairport-sync) the same track can be played without error so it is not specific to the data being sent.

ormanya, 2017-02-02:
Reading the issues over at https://github.com/mikebrady/shairport-sync, updating the version of shairport-sync used by Volumio might resolve the issue.

Nutcasey, 2017-03-25:
I have been seening that exact problem too wtih the same output.

closed 2017-06-02 by volumio