volumio/Volumio2 #991: Use Volumio as Bluetooth speaker

Octolinger, 2017-02-04:
By using a USB-Bluetooth adapter or a RPi3 connect a phone/tablet/computer and use Volumio as a Bluetooth speaker. You know, for us with Android phones and needs to stream music from our phones to bigger speakers.

chsims1, 2017-02-04:
Could you use an 'airplay' app connecting via wifi??

Octolinger, 2017-02-04:
Could, but then only local audio files. I want to cast all sound without voiding my phones warranty by rooting it.

xipmix, 2017-02-06:
This appears to be a duplicate, see #795.

Octolinger, 2017-02-06:
Yeah, it is. Did not find anything when I searched for it so I created a new one. Still, this feature should be possible to implement natively in Volumio (with working media controls and song info)

closed 2017-02-20 by volumio