volumio/Volumio2 #992: Language strings cause Volumio crash when uninstalling a plugin

tim-felix-lange, 2017-02-04:
I installed the GPIO-Buttons Plugin:
(Sidenote: When Installation was finished, the bar sprung back from 100% to 70% and was stuck, I had to refresh the page) 
When I click uninstall of the deactivated plugin, Volumio crashes with the following log: 

> Feb 04 21:56:27 volumio volumio[1211]: /volumio/app/index.js:1286
Feb 04 21:56:27 volumio volumio[1211]: if(this.i18nStrings[splitted[0]][splitted[1]]!==undefined)
Feb 04 21:56:27 volumio volumio[1211]: ^
Feb 04 21:56:27 volumio volumio[1211]: TypeError: Cannot read property 'CONFIRM_PLUGIN_UNINSTALL' of undefined
Feb 04 21:56:27 volumio volumio[1211]: at CoreCommandRouter.getI18nString (/volumio/app/index.js:1286:41)
Feb 04 21:56:27 volumio volumio[1211]: at Socket.<anonymous> (/volumio/app/plugins/user_interface/websocket/index.js:1488:81)
Feb 04 21:56:27 volumio volumio[1211]: at emitTwo (events.js:106:13)
Feb 04 21:56:27 volumio volumio[1211]: at Socket.emit (events.js:191:7)
Feb 04 21:56:27 volumio volumio[1211]: at Socket.onevent (/volumio/node_modules/socket.io/lib/socket.js:348:8)
Feb 04 21:56:27 volumio volumio[1211]: at Socket.onpacket (/volumio/node_modules/socket.io/lib/socket.js:308:12)
Feb 04 21:56:27 volumio volumio[1211]: at /volumio/node_modules/socket.io/lib/client.js:195:16
Feb 04 21:56:27 volumio volumio[1211]: at _combinedTickCallback (internal/process/next_tick.js:67:7)
Feb 04 21:56:27 volumio volumio[1211]: at process._tickCallback (internal/process/next_tick.js:98:9)
Feb 04 21:56:27 volumio mpd[594]: client: [2] closed

Uninstallation works fine if I set my language back to english, rather than german.

Vigo0101, 2017-02-09:
Hello,
I am experiencing a similar issue but when installing touchscreen or spotify plugins.
Switching back language to English before proceeding avoid trouble.

tim-felix-lange, 2017-02-13:
@Vigo0101 the problem seems to be independent of the plugin. As the error message already suggests, the string for "Do you really want to uninstall" is missing. This by itself is not a big problem, the problem is more that Volumio is not catching this error to fall back to english.

PR #1014 (2017-02-17): Avoid referencing undefined elements in i18nStrings array
Not all translations are complete, so we can't rely on i18nStrings having
even the top level of the language strings. To avoid crashing with a TypeError,
check the first element of a two-element template exists before trying to
check the second element exists.

Fixes: #992
closed 2017-02-17 by volumio