volumio/Volumio2 #993: Autoresize

mathieuroche, 2017-02-04:
Hello,

It's seems the auto-resize is broken.
https://volumio.org/forum/resize-partition-volumio-t4223.html

It's what i get on my rpi2 with a 16Gb card after a reboot.
volumio@volumio:~$ df
Filesystem                            Size  Used Avail Use% Mounted on
/dev/mmcblk0p2                        2.2G  521M  1.5G  26% /imgpart
/dev/loop0                            259M  259M     0 100% /static
overlay                                14G   17M   13G   1% /
devtmpfs                              502M     0  502M   0% /dev
tmpfs                                 511M     0  511M   0% /dev/shm
tmpfs                                 511M  9.1M  502M   2% /run
tmpfs                                 5.3M  4.1k  5.3M   1% /run/lock
tmpfs                                 511M     0  511M   0% /sys/fs/cgroup
tmpfs                                 511M   17k  511M   1% /tmp
tmpfs                                 511M     0  511M   0% /var/spool/cups
tmpfs                                 511M   46k  511M   1% /var/log
tmpfs                                 511M     0  511M   0% /var/spool/cups/tmp
/dev/mmcblk0p1                         64M   30M   35M  47% /boot
tmpfs                                 103M     0  103M   0% /run/user/1000

I try with fdisk (like here https://volumio.org/forum/using-extra-space-card-t2169.html) on the rpi2.
I also try with gparted on my debian.

But after, the rpi2 never reboot (no ssh access, no http access, ping OK) :/
Any idea ?

Thanks.

chsims1, 2017-02-05:
Volumio automatically resizes the partitions to fill remaining space on your SD card,  Give it time to do it's business.

xipmix, 2017-02-06:
What is supposed to get resized is the /data partition (/dev/mmcblk0p3). I don't see any /data partition mounted... so the partition may be getting resized, but not mounted. Can you post the output of
```
sudo parted -s /dev/mmcblk0 unit MB print free
```

xipmix, 2017-02-07:
Meh. I looked closer at what happens to /dev/mmcblk0p3 after the resize and now I see that partition is used as part of a union mount to make up the / filesystem (a readonly part and a read-write part).

I'm not sure what you see that is wrong here. There are other partitions on the card that use up part of it, so the '/' partition will never be 16Gb. Also, manufacturers ... choose their units carefully. Here is what I see on a pi2 running volumio (2.041) on an '8GB' card.
```
$ df -Ph
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2  2.0G  497M  1.4G  26% /imgpart
/dev/loop0      247M  247M     0 100% /static
overlay         5.1G   16M  4.8G   1% /
devtmpfs        479M     0  479M   0% /dev
tmpfs           487M     0  487M   0% /dev/shm
tmpfs           487M  8.7M  478M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           487M     0  487M   0% /sys/fs/cgroup
tmpfs           487M   16K  487M   1% /tmp
tmpfs           487M     0  487M   0% /var/spool/cups
tmpfs           487M   24K  487M   1% /var/log
tmpfs           487M     0  487M   0% /var/spool/cups/tmp
/dev/mmcblk0p1   61M   29M   33M  47% /boot
tmpfs            98M     0   98M   0% /run/user/1000

$ grep overlay /proc/mounts
overlay / overlay rw,relatime,lowerdir=/mnt/static,upperdir=/mnt/ext/dyn,workdir=/mnt/ext/work 0 0

$ sudo parted -s /dev/mmcblk0 unit MB print free
Model: SD SU08G (sd/mmc)
Disk /dev/mmcblk0: 7948MB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 

Number  Start   End     Size    Type     File system  Flags
        0.00MB  0.00MB  0.00MB           Free Space
 1      0.00MB  64.0MB  64.0MB  primary  fat16        boot, lba
 2      64.0MB  2300MB  2236MB  primary  ext4
 3      2300MB  7948MB  5648MB  primary  ext4
        7948MB  7948MB  0.21MB           Free Space

```
Notice the 2Gb taken up by the /imgpart partition. Yours is a little larger, but I think the auto-resize is working as expected.

closed 2017-02-20 by volumio

mathieuroche, 2017-03-06:
OK. I think I understand.
I believed that I must have a automount /data partition.
Sorry for the delay. AFK due to holiday ;)

df -h
```
Filesystem                            Size  Used Avail Use% Mounted on
/dev/mmcblk0p2                        2.0G  497M  1.4G  26% /imgpart
/dev/loop0                            247M  247M     0 100% /static
overlay                                13G   17M   12G   1% /
devtmpfs                              479M     0  479M   0% /dev
tmpfs                                 487M     0  487M   0% /dev/shm
tmpfs                                 487M   13M  474M   3% /run
tmpfs                                 5.0M  4.0K  5.0M   1% /run/lock
tmpfs                                 487M     0  487M   0% /sys/fs/cgroup
tmpfs                                 487M   16K  487M   1% /tmp
tmpfs                                 487M     0  487M   0% /var/spool/cups
tmpfs                                 487M  568K  487M   1% /var/log
tmpfs                                 487M     0  487M   0% /var/spool/cups/tmp
/dev/mmcblk0p1                         61M   29M   33M  47% /boot
//192.168.1.15/Multimedia/musique/hq  1.8T  1.8T   52G  98% /mnt/NAS/NAS
tmpfs                                  98M     0   98M   0% /run/user/1000
```
sudo parted -s /dev/mmcblk0 unit MB print free

```
Model: SD SL16G (sd/mmc)
Disk /dev/mmcblk0: 15932MB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags:

Number  Start   End      Size     Type     File system  Flags
        0.00MB  0.00MB   0.00MB            Free Space
 1      0.00MB  64.0MB   64.0MB   primary  fat16        boot, lba
 2      64.0MB  2300MB   2236MB   primary  ext4
 3      2300MB  15932MB  13632MB  primary  ext4

```

I don't know overlayfs so maybe I missed something but where is the setting of /dev/mmcblk0p3 ?

Thank again for your work on volumio.