volumio/Volumio2 #998: Version 2.041 does not open the Ethernet port on Raspberry Pi2

oldboybrown, 2017-02-05:
I have a 503HTA tube amp which works OK with a Pi3.  If I try to use the same setup using a Pi2 model B v1.2 there is no Ethernet connection (the LEDs stay off).  The Pi2 console also does not show using an HDMI monitor and keyboard (the Pi3 does).  The Pi2 works perfectly with Rasbian PIXEL.  I have tried these Volumio tests on both the Pi2 and Pi3, with and without the 503HTA attached to the Raspberry Pi, and the results are the same.  I am using a 3A power supply when the 503HTA is not attached.

I want to use the Pi2 to avoid the low level background noise issue on the Pi3.

Do you know what is causing this Ethernet/startup issue?

xipmix, 2017-02-07:
I tried 2.041 with a pi2 model B v1.1 (reading the version off the circuit board) and the ethernet has always worked fine.

Did you try [these tricks](https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=34061) to see if that gets the monitor working? volumio 2.041 is using kernel 4.4.9-v7+, what kernel is the PIXEL distribution using?

oldboybrown, 2017-02-10:
The kernel version is 4.4.9-v7+.

Sorry for the delay - I have been away.

oldboybrown, 2017-02-10:
By the way, the monitor works fine, but the Ethernet interface stays off.  This makes remote access impossible (SSH or browser).

xipmix, 2017-02-10:
Do you see the device in ifconfig?
```
$ /sbin/ifconfig
eth0      Link encap:Ethernet  HWaddr b8:27:eb:90:0d:21  
          inet addr:10.2.3.4  Bcast:10.2.3.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1260 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1601 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:102254 (99.8 KiB)  TX bytes:1792476 (1.7 MiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:718 errors:0 dropped:0 overruns:0 frame:0
          TX packets:718 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:67895 (66.3 KiB)  TX bytes:67895 (66.3 KiB)

```
Or lsusb?
```
Bus 001 Device 004: ID 0b05:17ba ASUSTek Computer, Inc. 
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
Also this is how dmesg looks for the network device (I included the boot line, it might matter)
```
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1200 bcm270
9.boardrev=0xa21041 bcm2709.serial=0x1b900d21 smsc95xx.macaddr=B8:27:EB:90:0D:21 bcm2708_fb.fbswap=1 bcm2709.uart_clock=3000000 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 vc_mem.mem_base=0x3ea00000 vc_mem.mem_size=0x3f000000  force_turbo=1 dwc_otg.lpm_enable=0  dwc_otg.fiq_enable=1 dwc_otg.fiq_fsm_enable=1 dwc_otg.fiq_fsm_mask=0x3 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 imgpart=/dev/mmcblk0p2 imgfile=/volumio_current.sqsh elevator=noop rootwait smsc95xx.turbo_mode=N
...
[    4.070991] smsc95xx v1.0.4
[    4.138191] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:90:be:eef

```

Can you compare the lsmod output with this, (2.041+pi2 B v1.1)
```
Module                  Size  Used by
nfsd                  246430  2 
cfg80211              427855  0 
rfkill                 16037  1 cfg80211
xt_REDIRECT             1340  1 
nf_nat_redirect         1454  1 xt_REDIRECT
xt_tcpudp               2269  1 
iptable_nat             1725  1 
nf_conntrack_ipv4      12912  1 
nf_defrag_ipv4          1428  1 nf_conntrack_ipv4
nf_nat_ipv4             4992  1 iptable_nat
nf_nat                 12375  2 nf_nat_redirect,nf_nat_ipv4
nf_conntrack           77388  3 nf_nat,nf_nat_ipv4,nf_conntrack_ipv4
ip_tables              11637  1 iptable_nat
x_tables               12507  3 ip_tables,xt_tcpudp,xt_REDIRECT
8192cu                519156  0 
bcm2835_gpiomem         3040  0 
bcm2835_wdt             3225  0 
uio_pdrv_genirq         3164  0 
uio                     8000  1 uio_pdrv_genirq
i2c_bcm2708             4770  0 
i2c_dev                 5859  0 
snd_bcm2835            20511  1 
snd_pcm                75698  1 snd_bcm2835
snd_seq                53119  0 
snd_seq_device          3621  1 snd_seq
snd_timer              19160  2 snd_pcm,snd_seq
snd                    51844  7 snd_bcm2835,snd_timer,snd_pcm,snd_seq,snd_seq_device
fuse                   83461  1 
ipv6                  347530  76 
squashfs               29079  1 
overlay                30212  1 
```
I also have a wifi usb thing but that's not configured, so it's not providing connectivity.

oldboybrown, 2017-02-11:
xipmix,

Many thanks for your help, but I cannot get far enough to execute these commands.  I have solved the problem by acquiring the same board as you are using (pi2 model B v1.1 ).  Using the same SD card, it just works.  I have no idea why the v1.2 won't initialise, but it behaved the same with two different Volumio downloaded SD cards, so I will put that down to experience and move on.  I have redeployed the model 1.2 in another project - so all is well.

Once again, thanks for putting in the effort to help.

closed 2017-02-11 by oldboybrown