volumio/Volumio2 #1007: APT sources on Raspberry Pi missing raspi.list

Gadgetoid, 2017-02-15:
I appreciate it may be intentional to avoid packages breaking Volumio, but I noticed on the Raspberry Pi that `/etc/apt/sources.list.d/raspi.list` is missing, and that `/etc/apt/sources.list` does not include `rpi` packages.

Is there a reason for this? While it's not difficult to add these sources manually, it would be nice to have easy access to packages such as `python-rpi.gpio` out of the box. They're useful to expand the functionality of a Pi-based Volumio build with hardware volume control, buttons, etc.

Thanks!

volumio, 2017-02-15:
Do you mean they should become: 
deb http://archive.raspberrypi.org/debian/ jessie main ui rpi ?

macmpi, 2017-02-15:
This was fixed recently in build as per https://github.com/volumio/Build/pull/116

closed 2017-02-20 by volumio

macmpi, 2017-02-21:
@Gadgetoid you may want to check this [plugin](https://github.com/tomatpasser/gpio-buttons) for GPIO related stuff.