volumio/Volumio2 #1016: Websocket API play command

Arthurin, 2017-02-18:
Quoting our beloved michelangelo (https://volumio.org/forum/websocket-home-automation-t4622.html) :

> (previous message)
```
socket.emit('play');
socket.emit('pause');)
```
> (michelangelo) 
> To be perfectly ok it should be 
```
> socket.emit('play', '');
> socket.emit('pause', '');
```

However socket.emit('play') does work while socket.emit('play', '') doesn't !
Is it possible to update the link in the doc for the Webscoket Interface ? It should be https://github.com/volumio/Volumio2/blob/master/app/plugins/user_interface/websocket/index.js and there we can see :
```
connWebSocket.on('play', function (N) {
				var timeStart = Date.now();
				if (N == null) {
					self.logStart('Client requests Volumio play')
						.then(self.commandRouter.volumioPlay.bind(self.commandRouter))
						.fail(self.pushError.bind(self))
						.done(function () {
							return self.logDone(timeStart);
						});
				} else if (N.value != undefined) {
                    self.logStart('Client requests Volumio play at index '+N.value)
                        .then(self.commandRouter.volumioPlay.bind(self.commandRouter,N.value))
                        .done(function () {
                            return self.logDone(timeStart);
                        });
				}
			});
```
So socket.emit('play', '') can't work (doesn't fill any of those 2 conditions).

Arthurin, 2017-05-15:
Another bug related to play position null : I noticed clearing Queue doesn't reset the position of the track number, it causes bugs with the play command, even in the webUI interface.
Let's say your current position in the queue is 10 and decide to clear the queue to make a new queue. Next (play,Null) will try to play a song at position 10. It's not the desired action (plus it'll raise an error if queue < 11 items)

volumio, 2017-05-15:
Interesting. Do you have any suggestion to fix the above behaviour?

Arthurin, 2017-05-15:
Current index should at least be reset to 0 when clearQueue is called.
To prevent similar errors (songs removed manually for example), a good scenario would be to be sure that current index is always < number of tracks in queue ;)

closed 2017-06-02 by volumio