volumio/Volumio2 #1025: Library scanning counts tracks twice with duplicate sources.

chsims1, 2017-02-22:
Not very important in the scheme of things, but I thought it was worth noting.  I set up a cifs and an nfs share of the same material on a NAS (I was troubleshooting), and Volumio reports double the expected unique tracks in the library. Presumably, this also means that any duplicate tracks on a single source will not be differentiated??

xipmix, 2017-02-24:
Do the shares have the same 'alias' in the MyMusic network drives listing? 
The URIs for the tracks include the mountpoint, it the two shares are mounted on different mountpoints this might explain what you see.

chsims1, 2017-02-24:
I'm almost positive that they were different alias', but the paths are obviously very different.

volumio, 2017-06-13:
Fixed

closed 2017-06-13 by volumio