volumio/Volumio2 #1031: Edit DNS via UI

macmpi, 2017-02-24:
Just a placeholder in reference to [this](https://github.com/volumio/Build/commit/077e503f73504dec0d602a98aa377313cfe3b0f9#commitcomment-20985929), so that some Network Preference Panel UI section is added to allow manual DNS entry: this may greatly help users having name resolution issues in specific cases (ISP filtering, bugs, etc).

macmpi, 2017-03-02:
Thanks @volumio for [this](https://github.com/volumio/Volumio2/commit/5fed4010f1c5d15df9f98e32cbd95a56707184c4). Closing.

macmpi, 2017-03-03:
A second thought: would be nice if current DNS values are displayed (like for IP addresses), as it is key to make a decision whether or not DNS are correct and need a change.

macmpi, 2017-03-04:
Actually I'll try to better express the requirement as a scrum-like user story.
As a user, I want to be able to see which are the actve DNS used by the system, and flip settings between automatic DHCP discovery, and custom manual entry.

macmpi, 2017-03-05:
@xipmix wrote:

> I think adding the custom nameserver lines to resolv.conf.head is better that adding to the resolv.conf.tail, but it would be nice to be able to opt out of using the google serversas well.

Yes.
Actually I think that with the new option to manually enter DNS info now, we can safely remove `resolv.conf.tail` altogether: this will indeed opt-out from Google servers, unless users positively decides to put them in manually.
`resolv.conf.tail` had been put-in initially as a workaround to fix some DNS issues, but it's a moot point now.

So let's:
- have manual entry as `resolv.conf.head`
- discard `resolv.conf.tail` from [Build script](https://github.com/volumio/Build/blob/63f44cb8f560ec3f64dbacba6536bb74d6756622/scripts/volumioconfig.sh#L432-L435) (note: OTA updated devices will keep it until reflash)
- have UI tweaks to show current DNS settings and allow to flip between auto and manual entry.

I wish I were more fluent in node and js so that I would do the PR myself...

macmpi, 2017-03-10:
Started to adapt to `resolv.conf.head` in https://github.com/volumio/Volumio2/pull/1061 PR
I realized that there is a critical UI issue currently **which may impact current users having DNS problems**:
- ON/OFF should activate or clear the .tail file (or move in/out with other name), and restart network
- SAVE should should not do anything to the file if OFF

Still need further UI work to display current DNS entries of active `/etc/resolv.conf` file

macmpi, 2017-03-20:
@volumio any chance you can have a look and fix the UI part, so that custom DNS can work fine in next release?

macmpi, 2017-04-26:
Custom DNS edition now fixed.

However keeping this issue open so that requirement for **UI Network Preference feedback on active DNS** ([as mentioned here](https://github.com/volumio/Volumio2/pull/1061#issuecomment-294122068)) does not go forgotten.

volumio, 2017-04-26:
Yes we will have it, no ETA now

macmpi, 2017-04-26:
Where is it better for tracking: here or in Volumio2-UI?

volumio, 2017-04-26:
here I would say

closed 2017-06-02 by volumio

macmpi, 2017-06-03:
renamed to capture the agreed feature request