volumio/Volumio2 #1033: Volumio2 web interface is not NAT-friendly

f18m, 2017-02-25:
Hi all,

I've been searching for solutions to this problem but so most of the stuff I found is related to Volumio 1.xx, so here's my question.

With volumio1 I was doing the following: I have an account on a dynamic DNS website, and then I configured my home router to forward port 80 from the public internet IP to the port 80 of the static IP I assign to volumio. It just worked fine.

I recently upgraded to Volumio 2.041 and I cannot access Volumio2 anymore from outside my LAN: if I access volumio from the LAN, everything works fine. If I access it via the dyndns address, I get and endless spinning wheel.

Using Google Chrome network inspection I see that was is failing is the HTTP GET to:
http://127.0.0.1:3000/socket.io/?EIO=3&transport=polling&t=LfIPnfI

Obviously the address "127.0.0.1" is wrong here and should instead be "mydyndnshost.changeip.org" or whatever.

I found a post with the suggestion of changing /volumio/http/restapi.js, line 49 from 
hostsarray.push('http://'+status.ipv4_address);
to
hostsarray.push('http://mydyndnshost.changeip.org');
but it does not work... 

I'm a developer (C++ not web actually) so can you point me to the place to look at? I could provide a tested patch that fixes this issue...

volumio, 2017-02-25:
Try this, /volumio/http/restapi.js line 7
change 
var primaryhost = undefined;

to 

var primaryhost = 'http://mydyndnshost.changeip.org';

before doing that, ensure you're on latest version by doing 

volumio pull

or better, download the new release (which I'm going to make available this weekend)

f18m, 2017-02-25:
Ok, great I will try. What about changing the port used by UI <-> backend communication?
At least with latest 2.041 configuring the router to forward port N!=80 to the port 80 of the Volumio2 static IP in the LAN did not work (at some point there are API requests that interrogate port 80)... 

A problem with exposing Volumio2 outside the LAN is that:
 - it does not have a login system
 - port 80 is a well known port and is one of the first attempts for "attacking" a public IP... 
For these reasons I would prefer to expose Volumio2 on my dynamic DNS but using a different port number... 

Thanks!

f18m, 2017-02-25:
I tried with "volumio pull" and changing primaryhost  and it works just fine with port 80... I get again network GET failures "http://127.0.0.1:3000/socket.io/?EIO=3&transport=polling&t=LfIPnfI" when I try to use a different port outside the LAN...

volumio, 2017-02-25:
Sorry but we designed the ports to work in just a single configuration.. You're on your own now, if you want to change port 80...

closed 2017-02-25 by volumio