volumio/Volumio2 #1034: Webradio Artist Incorrectly displayed.

chsims1, 2017-02-27:
When playing a webradio (v2.099), the first track shows the correct title and artist in the UI.  On subsequent tracks, the track title is changed correctly, but the artist name is not updated even though the correct info is in STATE SERVICE

`STATE SERVICE {"status":"play","position":0,"seek":615635,"duration":0,"samplerate":"22.05 KHz","bitdepth":"NaN bit","channels":2,"random":false,"updatedb":false,"repeat":false,"isStreaming":false,"title":"Just Like Heaven","artist":"The Cure","album":null,"uri":"http://icecast.timlradio.co.uk/vr32.ogg","trackType":"ogg"}`

cgtobi, 2017-03-01:
This is a duplicate of #819.

chsims1, 2017-03-01:
No, it isn't. #819 was for the track title, and appears to be resolved.  This issue is for the artist name.

cgtobi, 2017-03-01:
I'm sorry. It just looked too familiar.

chsims1, 2017-03-06:
V2.114 update: now working correctly ..ty

closed 2017-03-06 by chsims1