volumio/Volumio2 #1035: No sound on Raspberry Pi B with Hifiberry DAC

7091lapS, 2017-02-27:
I'm not getting any sound after mounting the Hifiberry DAC rca ([this one](https://www.hifiberry.com/shop/boards/hifiberry-dac-rca/)) on my Raspberry Pi B. Volumio is correctly configured to use the correct DAC, and I have tried rebooting multiple times.
Any help would be greatly appreciated.

gonace, 2017-03-05:
I've had the same problem since 22 Dec, I've downgraded (reinstalled) with a previous version and it seems to work.  (#927)  Tried 2.114 today and that didn't work either, so downgrading to a version before 2.041.

damien122, 2017-04-09:
I had the same problem with raspberry pi 2 and hifyberry DAC+.
I had fix this with a reboot after the next day. As I understand sometimes volumio cant load the hifyberry board. Only if you go under the play settings and there is a third sub menü with some specific hifyberry settings then the board was loaded successful and can play sound.

closed 2018-03-26 by volumio