volumio/Volumio2 #1037: Volumio logo kerning …

fooness, 2017-02-28:
Dear Volumio community, I think the kerning in Volumio’s logo can be improved very much.

I haven’t found the original logo file, but I made a quick and dirty preview with the logo from README.md …

With improved kerning—which is individual spacing between single characters (here: letters)—the logo has improved legibility and is much easier on the eyes.

If you want to use an improved version of your logo in the future, you could point me to the original logo file and I’ll do my very best to kern it ideally.

(Also, I think the ® is not really positioned good there, I would move it to the right and maybe even make it heavier, but I see why you did it like that, because it’s perfectly aligned with the outline of the o above.)

[image]

[image]

[image]

volumio, 2017-02-28:
You know what: you're absolutely right! It works really better! 
I am interested also in your solution for the r.
What else would you improve? 

Attached the most HI-res version we have. Is that too much if I ask yu a white version ? ( for dark backgrounds)... And feel free to propose anything you think would make it better! 
[image]

fooness, 2017-02-28:
Don’t you have like a vector file which you are using for all the different resolutions?

Else I’ll try to create one from this high resolution image.

Even better would be to know the typeface used here, if it’s not custom.

Then I could easily create the vector files for you.

Could you please give me the foreground colors (like the greyish/blackish one and the almost white one, plus the green color for the inside the o's)?

Thank you very much!

volumio, 2017-02-28:
Found the svg 👍 

[Volumio_logo_[registered]_20160713 V_modified.zip](https://github.com/volumio/Volumio2/files/809169/Volumio_logo_.registered._20160713.V_modified.zip)

We already added lots of svg files to Volumio's UI..
Do you still need the colours ?

fooness, 2017-02-28:
So, it seems the SVG is already pretty improved. Good work!

I just think I haven’t seen it in use, all the volumio logo appearances I remember had the kerning I posted above.

So maybe just replace the current logo PNGs/JPGs with new ones created from the SVG and then everything is superb.

Whoever created the SVG, maybe she or he can also create a white one, for compatibility reasons … though, I created a white (#fefefe) version, too, with Illustrator CC, SVG 1.1 compatible. I can also create others, if you want, but it seems the person who did the SVG is knowing what she’s/he’s doing!

Good luck in replacing all the used images files. : )

[volumio-svg-fefefe.zip](https://github.com/volumio/Volumio2/files/809260/volumio-svg-fefefe.zip)

whatsnewsisyphus, 2017-10-09:
I improved it a little further, pulling the first v to the left, aligning the green bits to be the same angle as the V's in the logo, and also kerning the text below.
[image]
[Volumio_logo_[registered]_20160713 V_modified_v2.zip](https://github.com/volumio/Volumio2/files/1367541/Volumio_logo_.registered._20160713.V_modified_v2.zip)

whatsnewsisyphus, 2017-10-10:
Let me know if you need anything

volumio, 2017-10-10:
You know what, I like it a lot! Do you mind sending a PR to our assets repo here?
https://github.com/volumio/volumio-graphic-resources

(see the logos folder)

whatsnewsisyphus, 2017-10-10:
No probs, will touch up a couple details and push it with all resolution assets. What about getting rid of the ®? It is kind of trite and rolling against the free software, copyleft goodness

volumio, 2017-10-10:
Let's keep the R for now (its a legal requirement since we registered the logo.... )

whatsnewsisyphus, 2017-10-10:
I'll do it later today. fyi: It is actually not a legal requirement, just
makes it easier to purse legal courses of action.

whatsnewsisyphus, 2017-10-11:
@volumio There are redundant resources in the logos folder, are they all being used? Are the given sizes optimal?

whatsnewsisyphus, 2017-11-01:
@volumio There are redundant resources in the logos folder, should I replicate them? Might be better if you weeded them out first.