volumio/Volumio2 #1045: ssh login seems to be broken after 2.114 installation

meve, 2017-03-04:
After updating the installation and restarting, logging in using ssh does not seem to be working any longer, even when using the right password:

```
$ ssh -o PreferredAuthentications=password -o PubkeyAuthentication=no volumio@volumio
volumio@volumio's password:
Connection closed by 192.168.1.2
```

xipmix, 2017-03-04:
Works fine here. Please check your known_hosts has the old host key removed
```
ssh-keygen -f ~/.ssh/known_hosts -R 192.168.1.2
```
Still not working? Can you set up screen & keyboard & login that way?

skikirkwood, 2017-03-04:
After upgrading to 2.114 ssh is broken for me too.

macmpi, 2017-03-04:
Is that OTA update or newly flashed image?
Works fine on my newly flashed image.

skikirkwood, 2017-03-04:
OTA update.  Let me download the latest and flash a new image to see if that fixes it.

chsims1, 2017-03-04:
OTA: couldn't authenticate over ssh
Flash: works fine

thomasdoerr, 2017-03-04:
Hi, I also update my two machine over the air and on both I'm unable to connect via ssh.
>>Connection to volumio-radio.local closed by remote host.

volumio, 2017-03-04:
From which version were you guys updating to?

xipmix, 2017-03-04:
I tested 2.114 download image, same as chsims1. Did not try OTA.

volumio, 2017-03-04:
I think I know what happened. I'm trying to replicate the issue here

chsims1, 2017-03-04:
Flashed new 2.041 to RPi0, updated to 2.114 via UI .... same ssh problem.

Log uploaded

volumio, 2017-03-04:
Ok, now please try to do : system -> delete user data 

if it does not reboot automatically (will take some 5 minutes at least), remove power and reapply. Can you tell me if that works from there?

chsims1, 2017-03-04:
Yes, clearing the 'user data' allowed me to log in via ssh.

volumio, 2017-03-04:
Ok. Good news is that after this update, the problem will not happen again.

thomasdoerr, 2017-03-05:
delete the user data allowed me to login too.

meve, 2017-03-06:
@volumio works for me too. Thanks!

heis2201, 2017-03-17:
What happens if one selects "Delete User Data". What exactly am I going to loose? My configs? Shares? Custom radio stations?

xipmix, 2017-03-18:
You should see, if press the button, some text like this:

    This will delete all your personal settings, favorites and playlists (USB and NAS music won't be
    deleted). The system will automatically restart once data has been deleted. Do you want to 
    continue?

heis2201, 2017-03-18:
So, that kind of workaround is not an option for me to get SSH working again. I should get familiar with local backups of Volumio2 - difficult without SSH. Hopefully a future update fixes the problem, will it?

xipmix, 2017-03-20:
The code behind that button does this
```
    fs.writeFile('/boot/user_data', ' ', function (err) {
        if (err) {
            self.logger.info('Cannot User Data delete file');
        } else {
            self.logger.info('Created User Data delete file, rebooting');
            self.commandRouter.reboot();
        }
```
ie it creates an empty file in /boot, then reboots. The file it creates does not exist on my system, so it seems to be created as a flag to the system when it next boots. After a bit of poking around I found that the Build project has an initramfs script that recreates the 'data' filesystem
```
mkdir /mnt/factory
mount -t vfat /dev/mmcblk0p1 /mnt/factory
...
if [ -e "/mnt/factory/user_data" ]; then
  echo "Deleting User Data"
  mkfs.ext4 -F -E stride=2,stripe-width=1024 -b 4096 /dev/mmcblk0p3 -L volumio_data
  rm /mnt/factory/user_data
  
  echo "User Data successfully deleted "
fi
umount /mnt/factory
```
So it really does what it says on the tin. Not much scope for partial deletion, either.

Backing up -
```
laptop$ rsync -av volumio@volumio:/data ./volumio-backup/
```
should do the trick, I don't think there are other areas where user data is stored.

So in principle you could back up, then delete user data and try to restore.
A safer approach would be flash a separate card and try to restore your data to that, keeping the old card safely tucked away. You may be able to get away with restoring all of /data, but most likely you will have to reflash and then restore one piece at a time so that SSH is not broken again by some piece of broken configuration. It could be a lengthy exercise.

heis2201, 2017-03-20:
Seems a like a lengthy process indeed. What's the reason for the broken ssh login? My preferred method would be to fix the cause (i can login locally to my rpi).

Will a future volumio update restore ssh login?

macmpi, 2017-03-22:
for ssh reset, you may just want to try this under console (this does not affect user data):
```
sudo rm -v /etc/ssh/ssh_host_*
sudo dpkg-reconfigure openssh-server
sudo shutdown -r now
```

arthurlutz, 2017-03-22:
affected by this too. hope the next OTA release fixes this automatically.

klins1bfh, 2017-03-30:
I just updated and now cant ssh to volumio anymore.

macmpi, 2017-03-30:
Please refer to already mentioned [responses](https://github.com/volumio/Volumio2/issues/1045#issuecomment-284157383).
If you want to do some manual back-up/restore for now, please check [this](https://github.com/volumio/Volumio2/issues/1048#issuecomment-288077276)

Anyhow, re-imaging is better for now.

dup2, 2017-05-01:
Any progress on this? Deleting all user data is certainly not an option for fixing a bug which was introduced by an OTA update..

Arthurin, 2017-05-17:
Warning ! "delete user data" doesn't delete only data related to volumio, but also all programs installed with apt-get and so on!

heis2201, 2017-06-27:
Is it possible that ssh access is broken again after an update to 2.201?

heis2201, 2017-06-27:
ssh access needs to be enabled first: https://volumio.github.io/docs/User_Manual/SSH.html

Sorry for jumping on this old issue here.

abenarroch, 2017-08-30:
enabling SSH in volumio.local/dev or 10.0.0.xx/dev worked great.

this would be a good candidate to move to the gui, for now I am happy that it worked.

many thanks

closed 2017-08-30 by volumio