volumio/Volumio2 #1047: No SMB mounts after 2.114 install

skikirkwood, 2017-03-04:
My two SMB mounts are not showing up after upgrading to 2.114

chsims1, 2017-03-04:
Seems to be working here:

`smbclient -L 192.168.2.222
Enter ian's password: 
Domain=[WORKGROUP] OS=[Windows 6.1] Server=[Samba 4.2.14-Debian]

        Sharename       Type      Comment
        ---------       ----      -------
        Internal Storage Disk      Volumio Internal Music Folder
        USB             Disk      Volumio Internal Music Folder
        NAS             Disk      Volumio Internal Music Folder
        IPC$            IPC       IPC Service (Volumio Audiophile Music Player)
Domain=[WORKGROUP] OS=[Windows 6.1] Server=[Samba 4.2.14-Debian]

        Server               Comment
        ---------            -------
        READYSHARE           readyshare
        VOLUMIO              Volumio Audiophile Music Player

        Workgroup            Master
        ---------            -------
        WORKGROUP  `

macmpi, 2017-03-04:
Probably same OTA vs new image issue?

xipmix, 2017-03-04:
Could you post some logs ... once ssh is going.

volumio, 2017-03-04:
or, better. 
Go to /dev page 

fill the send log or bug report with your issue, click send and paste here the link

skikirkwood, 2017-03-04:
http://logs.volumio.org/volumio/aufovCU.html <http://logs.volumio.org/volumio/aufovCU.html>

skikirkwood, 2017-03-05:
Fresh install (vs. OTA) fixes this issue.

2lepus, 2017-03-05:
I saw the old entries, but they all were unmounted.
I tried to edit the entry or mount other shares without success.
But I found out that I could mount the root share.
i tried it with deleting user data (#1045) and then I was able to mount them all again.

jaui, 2017-03-08:
@skikirkwood
please don't use upload volumio logs function since the log contains plain text password for the used spotify account. please change your spotify password.

closed 2017-06-02 by volumio