volumio/Volumio2 #1048: User data migration

macmpi, 2017-03-04:
As a user I want to be able to simply migrate my data from one Volumio instance to another one.

This can be useful to migrate all share settings, preferred radios, eventually plugins, etc.. between several installs, or ease migration when starting "fresh" from a new image.
Of course in that latter case there may be incompatibilities, but in such case user can always wipe all data if it does not work properly.

One simple method for user would be a user datat export/import button on system UI that downloads to/uploads from PC a compressed archive of /data (except eventual swapfile).

chsims1, 2017-03-04:
Yes, this would be very useful.  Requested as plugin on forums, but I think it should be an essential part of the UI.

skikirkwood, 2017-03-05:
We've started a MyVolumio project that will support most of this.  I have an initial prototype built using Drupal, which uses Volumio's RESTful API to request playlists, favourites, config settings, etc. and store them in the Drupal DB.  Restoring works as well.

Lots of UI work to do before we can put an alpha out, but it's coming.  We'll also use this to let people do Oauth authentication with systems like Spotify.   With that available we'll be able to implement the next phase of Spotify integration accessing people's saved albums, tracks, artists, etc. and provide an experience closer to the Spotify Web player.

macmpi, 2017-03-05:
Thanks for the info, very interesting indeed.
Will that project be included by default into Volumio, or will that be an 3rd party add-on module?

skikirkwood, 2017-03-05:
MyVolumio will be a web site, and Volumio will include a connector to it.  Exactly how it will be structured and rolled out is not defined, and Michelangelo will be figuring that out.

Good news is a local dev version let me restore my web radios, playlists and config to a newly flashed Volumio Pi after, and it was pretty exciting see that!

You can see an early dev version here:

http://eejambrasdev.prod.acquia-sites.com/

Notice we have multi-lingual support, with automated machine translation of all new content thanks to Lingotek.

If you browse the [Volumio Backups](http://eejambrasdev.prod.acquia-sites.com/volumio-backups) tab and click on the any of the entries you will see the Json content that came from a REST call to Volumio.  The Restore tab doesn't have a working UI yet, but plugging in the right URL will restore any stored setting back to your Volumio machine.

Besides the UI we need to support multiple Volumio devices, and authentication, and a few other things.  But good to know the core plumbing is already in place!

The Github repo for the work is here:

https://github.com/skikirkwood/MyVolumio

macmpi, 2017-03-05:
Very interesting.
Surprised this comes as a web service though, would have anticipated such kind of feature would be local.

Is there a plan to attach Volumio to a service offering, or munge some user data, do analytics, etc?

skikirkwood, 2017-03-05:
As a web service we can do a suite of new capabilities.  Have leaderboards of most popular albums/tracks/artists among Volumio users, sharing of web radio favourites, etc. that could not be done easily with a local only service.  

Lot of possibilities of what can be done, but to start step one will be backing up and restoring Volumio settings/data among multiple devices.

macmpi, 2017-03-05:
Sure: lot's of stuff can then be done remotely, even administering local players (updates, etc), content suggestions push, up to monetizing intelligence to music services, data usage monitoring, etc...
I incidentally noticed a recent [google analytics](https://github.com/volumio/Volumio2-UI/commit/9e2ca4b2a08a376732ed89faf4435ff105b7e078#diff-e249faefed5757034596c5096d33dab6)-related submit which may illustrate such directions.

However, I'm not sure all users will be keen on sharing personal info like music collection, preferred tunes across music services, data usage, etc. Hopefully users will be able to opt-in/out easily with some great level of granularity. Regional regulations on personal info management is also going to be a challenge to manage.
I hope Volumio will still keep its great focus on doing stuff right locally for most, without enforcing web-service connection (which I agree can still provide great value for many).

Thanks a lot for sharing the vision of future enhancements.

volumio, 2017-03-05:
Yes our main focus will still be local features, but we'll add myvolumio as a premium service, only for user wanting to opt in. The idea is to build a cloud-service that will extend current volumio functionality. The goal is to get a constant revenue stream to fund the development of the open volumio echosystem, and to build a state of the art music-enjoyment service...
I will share more details on it this week. The GA commit you saw is just for fun, but we're working on some really cool stuff on the cloud side... Few days and you'll know more

skikirkwood, 2017-03-05:
Yes, absolutely everything I mentioned would be an opt-in.  Again, step one is to get storing/restoring of settings among multiple Volumio devices, followed quickly with the ability to authenticate with Oauth-based services like Spotify.  We can’t actually get any further with the new Spotify Web API until we have Oauth authentication.

I also think it will be easier to support a range of capabilities through a centralized server over having every Volumio user run some kind of web-based server locally in addition to their instances of Volumio on their devices.  I was an early user of the Squeezebox Touch.  Despite being a computer scientist, I had numerous issues with operating the Logitech Media Server on my Mac that required I called into Logitech support.  Their support costs are probably the reason behind discontinuing the Squeezebox line.

That said, having a web service that Volumio users and their devices can interoperate with will open a new world of possibilities.  Should be a lot of fun!

chsims1, 2017-03-06:
This all looks very exciting, with loads of possibilities, but I still think that a basic save/restore settings from the Volumio UI is very desirable.  Is this not simply a case of saving various config files, or am I being too simplistic?

gkkpch, 2017-03-07:
Very interesting indeed!! A major next step in Volumio's history

macmpi, 2017-03-08:
Back to the original topic, the recent [log issue](https://github.com/volumio/Volumio2/issues/1047#issuecomment-284928779) shows how sensitive information can leak very easily (this one is fixed now).
Such example would plea for **back/restore to be a local-only feature**, as there is high risk and very little value to upload, for instance, all music services account/password info (which may indirectly link to banking info themselves) to myVolumio cloud service.

skikirkwood, 2017-03-09:
Hi @macmpi, totally agree we should not be backing up unencrypted account/password in the Cloud!  We can follow the Apple iPhone backup model.  You can backup to the Cloud and have a password to encrypt all info, or not have a password and account/passwords will be stripped out of the backup.

I also think it's a great idea to have a local backup/restore feature.  That said, I personally enjoy developing Cloud-based technology, and I think there's a world of capabilities we can provide to the Volumio community that can only be accomplished with a Cloud-based platform.

macmpi, 2017-03-09:
@skikirkwood wrote:
> I also think it's a great idea to have a local backup/restore feature.

Great. Since you already developed working code for this in the MyVolumio framework, any chance you may adapt it for 100% local operations as described initially and release such version separately?
That would be of great value for all users!

skikirkwood, 2017-03-10:
@macmpi, the issue is how would we deliver the code locally.  Since what I prototyped was a web site, it would be daunting for people to install locally an entire Drupal web site, running Apache, with a MySQL database, ....  

Having a simple local desktop app seems to make more sense, but then we would need versions for Windows, Mac and Linux.  I don't see that happening.  It's possible you could build a cross-platform app in Java or Flex, but those things have a host of problems themselves.  

So I still go back to a web service....

macmpi, 2017-03-10:
Ok, so unfortunately one would probably better starting from scratch to code such local import/export data, as the only real interaction needed with PC/Mac/Linux in such case is really just to upload/download an archive through Volumio the web interface.

So, let's keep this issue open for now I guess, so that such parallel track is considered.
Thanks.

skikirkwood, 2017-03-10:
Yes, I think that makes sense.       

> Ok, so unfortunately one would probably better starting from scratch to code such local import/export data, as the only real interaction needed with PC/Mac/Linux in such case is really just to [...]

 ]]>

skikirkwood, 2017-03-13:
I've been really busy and didn't make it a priority to change my Spotify password.  So I'm playing Spotify tonight and it started stopping with a message that a stream was playing elsewhere.  I restarted my playlist and got into a loop with whoever was using my account.

So uploading the log did cause someone to hack my account, and then they changed my password.  This is why Oauth is so superior over storing your userid and password in plain text in a site.

Ok, I learned my lesson.  And more motivated than ever to replace the current Spotify plugin with a 100% use of the new Spotify Web API that uses OAuth.

volumio, 2017-03-13:
Jim, that's a serious issue. I am removing all the logs now. Then I'll check if there are still residuals of private datas .Sorry for that

skikirkwood, 2017-03-13:
It's even worse!  Whoever hijacked my Spotify account also tried to break into my online bank account and tripped the max # of password attempts.  Good thing I use 1Password to generate unique passwords for all sites I use!

Odd thing is I had a unique email address for Spotify, so no idea how they associated my main email address with the bank account.  And good thing my bank requires TFA.

volumio, 2017-03-13:
Maybe volumio was not the entry point then... Is there something I can do to help?

skikirkwood, 2017-03-13:
I'm good. Spotify has great tech support via chat. Just had to show a receipt from a bill and they took care of it immediately.       

>   
>   
>   
>   
>
> Maybe volumio was not the entry point then... Is there something I can do to help?

 ]]>

macmpi, 2017-03-13:
cool, what a relief!

volumio, 2017-03-13:
Yes, definetely...

BorisHollas, 2017-03-20:
Why can't there be separate partitions for / and /home with all user data store in /home? If I update my Ubuntu, my /home remains untouched, so my settings aren't erased. The user data can then be migrated by copying the /home partition.

volumio, 2017-03-21:
Instead of copying /home you copy /data and all is fine

macmpi, 2017-03-21:
Wouldn't this generate awkward situations with plugins though, as they will appear as somehow installed on new image with such migrated /data, but will missing all the eventual linux packages dependencies...
So to be safer, you might want to just .tgz the following to some location (/boot for instance):
- /data/INTERNAL/*  
- /data/albumart/*
- /data/backgrounds/ (only your eventual custom ones)  one idea: maybe volumio's base could be in separate folder to help discriminate...
- eventually (but more "risky") /data/configuration/* while excluding plugins.json (but it may conflict if new image change some configuration related stuff)
- /data/favourites/*
- /data/playlist/*
- /data/queue

And then re-expend on new image once booted.
Then re-install the plugins (if you decided to move-in configuration folder, your plugin will most likely enjoy all previously set options)

Do I get it right?

xipmix, 2017-03-22:
Just noting #1045 is related to this.

arthurlutz, 2017-03-22:
@macmpi by the way I also noticed the google analytics tracker and added an issue here https://github.com/volumio/Volumio2-UI/issues/310

macmpi, 2017-07-11:
For those interested, took a first stab at a plugin: feel free to [test it](https://github.com/macmpi/volumio-plugins/raw/gh-pages/plugins/volumio/armhf/system_controller/backup_restore/backup_restore.zip) and report eventual issues [there](https://github.com/macmpi/volumio-plugins/issues)
For now, archive is generated in root of SD fat partition, and is used from there for restore.
Hope it helps.

xipmix, 2017-07-11:
Took a look at the code, looks quite plausible to my inexpert eye. Haven't tested yet. I like how you break the save into separate steps for each chunk, and leave the (large?) albumart to the last. 

Why do you delete  ```data/configuration/plugins.json``` - because the plugins aren't backed up and must be reinstalled?

As I read, I kept thinking 'what if /boot fills up?'. It might be a little more robust to collate the tarball in e.g. /data/backup_and_restore_temp/ and gzip it, then move it to /boot.  I don't have a feel for how much of an issue this is likely to be. I have 28M (45%) free on /boot.

For the paranoid, it might be worth adding a check for how large the albumart is and showing the size in a popup if it is "large", ie more than some fraction of the free space on /boot. Or if the albumart is going to make the tarball too big, offer the user the option of not including it. Since the albumart is mostly jpeg (ie incompressible) 'du' would give a reasonable estimate that could be made before adding it to the tarball.

chsims1, 2017-07-11:
Good start I think.
Couple of points:
(1) restore fails ... permissions?
(2) save space by having backup and restore buttons together?  I was initially puzzled as to how to restore when viewing on a small screen.

macmpi, 2017-07-11:
All good comments thanks. Please [open issues in my repo](https://github.com/macmpi/volumio-plugins/issues) for follow-up.
To avoid overflowing this thread, I'm closing this particular issue now: thanks!

closed 2017-07-11 by macmpi