volumio/Volumio2 #1055: Random playback stops after a few stops (randomly)

gkkpch, 2017-03-07:
I thought I noticed this on an earlier C1 release but never paid much attention to it.
It seems more users experience this and this is not a board or os issue.

To reproduce, add a number of albums to the queue, start random play and wait. 
It takes a couple of songs (number is random) and then it playing stops and the end of a song

biva, 2017-03-15:
Hello, I'm experiencing this also. Any news?
Available for testing if needed.

gkkpch, 2017-03-17:
@biva: no news yet. This might take a while, so please bear with us.

marianol, 2017-03-22:
Same thing here (RasPI 2 / Volumio 2.118) Queue in Random and repeat. Testing again with the list without random or repeat and will post the results.

marianol, 2017-03-22:
OK if I disable random music keeps playing.

Chaz-be, 2017-03-26:
@gkkpch I have the same problem on raspberry pi 2 and 3 (also with clean install V2.118 ) with or without a DAC, on random list. 

with the previous version  I have the same problem

with the 2.041 no problem on all of my raspberry 2 and 3.
I return to the 2.041 on each raspi for the moment. :s

Chaz-be, 2017-03-27:
Available for testing if needed.

RastaX, 2017-05-07:
This was claimed to be fixed in 2.129:

> 2.129 (25-03-2017)
> FIXES 
> Fixed random stop of playback

But the problem is still present on current v2.129 (26-03-2017) on Odroid C2.

gkkpch, 2017-05-07:
Odroid and other armv7 devices have not been released yet, pending mpd compilation. They will follow very soon.