volumio/Volumio2 #1067: Volumio doesn't work with no internet

Gridgway1960, 2017-03-13:
When setting up a completely private volumio on R-Pi in an environment where there is no internet (no DNS etc) you cannot select music to play.  If you select artists, the artists are listed, but if you click an artist to see the albums, nothing happens.  The setup is R-Pi, TP-Link wireless router plus a QNAP NAS.

fanciulli, 2017-03-15:
Hello @Gridgway1960 ,
i was testing this issue but couldn't replicate it . I had disabled the DNS to avoid any internet connection but seems that my pi is working well.
Do you have additional info that could drive me to replicate the issue?

Gridgway1960, 2017-03-15:
I'll get some more info on exact IP settings from the network this evening if that might help @fanciulli ?

Gridgway1960, 2017-03-15:
Hmmm, not sure I have got anything really useful to add.

If I am connected to the network (that doesn't have internet) on a PC and ping www.google.com I get "ping request could not find host"

If I tracert www.google.com I get "unable to resolve target system name"

If I ping 8.8.8.8 I get "Destination net unreachable"

This is all on a net with 192.168.43.1 as the router address which is also the DNS address.  Is it possible that different routers return different errors when they don't have access to a DNS?

I can log in to the R-Pi and run some commands if that's helpful, but I'll need to know what you'd like to run.  I'm a linux noob, so you'll have to tell me the exact commands to run!!

HTH
Graham

PS as some

closed 2017-06-02 by volumio

Gridgway1960, 2017-06-03:
Does that mean the issue is fixed in a new release?  Thanks!