volumio/Volumio2 #1074: Periodic "My Music" update

mattberther, 2017-03-17:
Perhaps I am missing something, but I would love to have some automated process that rescans (actually, I think the volumio terminology is updates) my music library (hosted on my NAS). Currently, I need to come in and manually click the update button.

Also, would love to see a list of "most recently added" music from the home screen. 

Thank you for considering.

closed 2017-06-02 by volumio

mattberther, 2017-06-08:
What was the resolution on this?

volumio, 2017-06-08:
Volumio should automatically update itself when new files are added. 
as for the recently added, we're working on it

mattberther, 2017-06-30:
I cant really seem to get this to work. I have a NAS setup as an NFS share. When I add music, unless I manually click update, they don't come in.

volumio, 2017-06-30:
It already does it every 4 seconds

mattberther, 2017-07-03:
I know you keep saying its supposed to work. But, it does not work for me. Are there logs that I can review somewhere to try to get to the bottom of this?

kehugter, 2017-10-21:
Permissions?
You ssh as `volumio`, who also owns the nfs mounts, that's why you can see all files. But the scanning is done by mpd, which runs as `mpd:audio`. Thus, if your dirs/files in the nfs mount don't have enough world permissions, they can't be accessed and indexed.

You could change your local permissions, but as far as I can tell (after little testing), mpd can run as `volumio:audio` to access the whole mount and nothing breaks. @volumio what do you think?

@mattberther if you want to try it out, edit `/etc/mpd.conf`

```
user "volumio"
```

and then

```
sudo systemctl restart mpd.service
```

mattberther, 2017-11-20:
Thanks for the thoughts, @kehugter. Unfortunately, no dice. I'm still not able to have it automatically rescan my library. I've even tried switching to a cifs share with no luck.

kehugter, 2017-11-21:
I'm pretty sure my library isn't updating the files on my NFS share every 4 seconds either. Any suggestion on how to debug this further, @volumio?

mattberther, 2017-11-21:
I *know* mine is not. I add new files; I can see them when I SSH; I can see them when I browse the NAS through the UI. But, unless I manually click update on the My Music tab, Im not able to search for it.

I've let the music sit for up to 3 days to see if it picks up. Ive also tried restarting. Nothing, except for clicking Update, makes the music searchable in the library.

volumio, 2017-11-21:
Hi Matt,
I've finally came to the conclusion that inotify is not reliable over SMB\NFS. Therefore the auto update via NAS does not work.
To overcome this, I've added the "update this folder" function in every folder, just click the 3 dots on your NAS or any subfolder and you will be able to just update this subfolder.

mattberther, 2017-11-21:
Huzzah! Thank you. That will work!

marcolino7, 2019-12-13:
Hi,
just quick question, is possible to schedule a complete library rebuild or simply an update via cron?
In this way i do not need to reload manually everytime I add a new folder.

Thanks