volumio/Volumio2 #1080: Enable support for Ralink MT7601U Wireless Adapter

heis2201, 2017-03-20:
Ralink **MT7601U** Wireless Adapter is supported by the current Linux kernel but somehow not supported by Volumio2. Can you please add the driver for this USB dongle in future releases.

Many thanks!

macmpi, 2017-03-20:
Maybe firmware is missing. You may want to try this:
`sudo wget -O /lib/firmware/mt7601u.bin https://github.com/porjo/mt7601u/tree/master/src/mcu/bin/MT7601.bin`

volumio, 2017-03-20:
Can we add it to the build script?

macmpi, 2017-03-20:
Sure, will do if solution confirmed.
Will do for all platforms I guess (volumioconfig.sh), not just Pi, right?

volumio, 2017-03-20:
Yes I would say

macmpi, 2017-03-20:
Will add a note that for kernels lower than 4.2 (may be the case for some other platforms?) a driver still needs to be compiled and added.

BTW, pls check Build PR for similar inclusion of Atheros firmware

heis2201, 2017-03-20:
Hi friends, many thanks for your prompt replies. I would love to test the solution but I am constrained by issue #1045. (I am not ready to *delete user data* and thus restore ssh access since I would loose my settings and custom playlists, and without ssh it seems hard to create a backup despite the fact that I wouldn't know what to backup. )

I may just connect a display and a keyboard and log in locally, and then test the solution. I'll keep you posted.

macmpi, 2017-03-20:
@heis2201 understand your ssh problem after OTA, and willingness for a fix to keep your data untouched.
However, by nature OTA and new image upgrades are different in nature and some system updates changes do require to go through re-imaging: it seems it's the case for this ssh issue and other base OS fixes in 2.118.
Understand the inconvenience but you may need to go over it.
Ideally, in a close future there will be some [data export/import tool](https://github.com/volumio/Volumio2/issues/1048) that will somehow help migration when re-imaging, but it's not the case yet.
In the meantime you may have to "take the pill" and manually rebuild your preferences on new image...If you don't, you may uncover other unexpected issues (there are many in forum), including additional ones which may pile-up with next OTAs...

BTW, and as an example, the change we talk about here (inclusion of MT7601U firmware), will require re-imaging to benefit from it from first boot...

heis2201, 2017-03-20:
@macmpi it's ok, the need for flashing is not the end of the world.

I just want to make it as simple as possible ...

- Where  can I read and understand what causes the denied ssh access? Is there no way to login locally and fix it?
- Is there a nice way to export my custom radio stations? I am wondering if I have to manually add them all, or if there is/will be an easy way (a plugin?) to import them as .pls/.m3u (as we get them from the radio stations) and as they were understood by volumio 1.5. #334 
- MT7601U firmware: I don't think it requires reflashing. One doesn't need the module to be loaded from the first boot since you have to use a LAN/ethernet cable on first boot to configure you system anyway, no? At the same time, one can download the driver and load the necessary kernel modules. Reboot and done.
- As for backup, please don't take it personally, but I am strongly opposed to a cloud-based backup of my personal data. I would instead very much prefer a simple local backup solution. I simple script that creates a tarball with all personal data, and the possibility to download that archive from the WebUI. I understand the need of a Premium service to generate some revenue for all the effort being put in here, no question. The cloud-based setup should just not be the only one.

Sorry, we are going slightly offtopic here. What really interest me is: Can I fix the ssh problem if I log in locally?

macmpi, 2017-03-20:
Indeed many of your questions would rather find answer in other relevant open issues I think.
As for back-up, I think you probably misread my position which calls for local solution: but feel free to comment there.

As for original issue, you may indeed be able to test proposed fix from console despite ssh issue.
If you get a chance to test it, can you also check if both wifi mode work: client to Home wifi AP, and Volumio a hotspot?
Thanks for your feedback.

macmpi, 2017-03-27:
@heis2201 , @shoonoise
Did you have a chance to test the [proposed solution](https://github.com/volumio/Volumio2/issues/1080#issuecomment-287686609) and verify if Home wifi AP connexion now works with your Ralink MT7601U dongle?
What about Volumio Hotspot? 

Thanks for feedback.

heis2201, 2017-03-30:
Sorry, I'm traveling, currently no time to test the solution proposed. 
I'll let you know when I get a chance to test it.

macmpi, 2017-03-30:
Sure, no worries.
It's parked until we hear about any user feedback.

macmpi, 2017-04-15:
@heis2201  @shoonoise
Any update on this: If no longer an issue, can you then close it?
If the proposed fix helped, please report so that we include in a next release.

macmpi, 2017-06-21:
@heis2201 any feedback on [proposed solution](https://github.com/volumio/Volumio2/issues/1080#issuecomment-287686609)?
Please close if no longer an issue

heis2201, 2017-06-27:
Sorry for delayed feedback here. I was going to try it now but I have no ssh access (again). I had that problem before, and its was introduced by an OTA update. Now again?
Anyone any idea what's going on? My release. 2.201.

volumio, 2017-06-27:
https://volumio.github.io/docs/User_Manual/SSH.html

heis2201, 2017-06-27:
@macmpi The firmware file used in the [proposed solution](https://github.com/volumio/Volumio2/issues/1080#issuecomment-287686609) is no longer available in that repository. I found a bunch of other [MT7601U repos on github](https://github.com/search?q=mt7601u). Any idea which of thise i could use?

heis2201, 2017-06-27:
I looked up /lib/firmware and mt7601u.bin appears there - it seems to be included with newer kernels. So, I tried a `sudo modprobe mt7601` which went fine. The webui doesn't show the adapter. Would it show a second adapter at all?

heis2201, 2017-06-27:
`lsusb`:

```
 *Bus 001 Device 007: ID 148f:7601 Ralink Technology, Corp.*
```

`lsmod`

```
Module                  Size  Used by
cpufreq_stats           4040  0 
xt_REDIRECT             1547  1 
nf_nat_redirect         1911  1 xt_REDIRECT
xt_tcpudp               2455  1 
iptable_nat             2224  1 
nf_conntrack_ipv4      14594  1 
nf_defrag_ipv4          1817  1 nf_conntrack_ipv4
nf_nat_ipv4             6306  1 iptable_nat
nf_nat                 17370  2 nf_nat_redirect,nf_nat_ipv4
nf_conntrack           97429  3 nf_nat,nf_nat_ipv4,nf_conntrack_ipv4
ip_tables              12473  1 iptable_nat
x_tables               18811  3 ip_tables,xt_tcpudp,xt_REDIRECT
nfsd                  263695  2 
mt7601u                86457  0 
mac80211              608549  1 mt7601u
cfg80211              499234  2 mac80211,mt7601u
rfkill                 21397  1 cfg80211
snd_soc_bcm2835_i2s     6994  2 
snd_soc_pcm5102a        2205  1 
snd_soc_hifiberry_dac     3027  0 
regmap_mmio             3728  1 snd_soc_bcm2835_i2s
sg                     20799  0 
snd_soc_core          164100  3 snd_soc_bcm2835_i2s,snd_soc_hifiberry_dac,snd_soc_pcm5102a
snd_bcm2835            23163  1 
snd_pcm_dmaengine       5906  1 snd_soc_core
snd_pcm                95441  4 snd_bcm2835,snd_soc_core,snd_soc_hifiberry_dac,snd_pcm_dmaengine
snd_seq                58506  0 
snd_seq_device          5266  1 snd_seq
snd_timer              22396  2 snd_pcm,snd_seq
snd                    68368  8 snd_bcm2835,snd_soc_core,snd_timer,snd_pcm,snd_seq,snd_seq_device
bcm2835_gpiomem         3823  0 
i2c_bcm2708             5740  0 
bcm2835_wdt             4133  0 
uio_pdrv_genirq         3718  0 
uio                    10230  1 uio_pdrv_genirq
i2c_dev                 6578  0 
fuse                   89800  1 
ipv6                  367575  60 
squashfs               32128  1 
overlay                32464  1
```

Interesting part of `dmesg`

```
mt7601u 1-1.4:1.0: ASIC revision: 76010001 MAC revision: 76010500
[   16.537238] mt7601u 1-1.4:1.0: firmware, attempted to load /lib/firmware/mt7601u.bin, but failed with error -22
[   16.552288] mt7601u 1-1.4:1.0: Direct firmware load for mt7601u.bin failed with error -22
[   16.569939] mt7601u: probe of 1-1.4:1.0 failed with error -22
[   16.581762] usbcore: registered new interface driver mt7601u
```
How to lookup error *-22*?

heis2201, 2017-06-27:
Looks like I don't have the firmware, the file is empty. No clue how that happened.

> -rw-r--r-- 1 root    root         0 Jun 27 17:22 mt7601u.bin

heis2201, 2017-06-27:
I am re-trying now with firmware from here: git.kernel.org/cgit/linux/kernel/git/firmware/linux-firmware.git/plain/mt7601u.bin

heis2201, 2017-06-27:
I would be nice if the firmware could be added to the distro by default.

macmpi, 2017-06-27:
Trying to find a standard package for it.
It seems it's in Debian's [firmware-misc-nonfree](https://packages.debian.org/sid/all/firmware-misc-nonfree/filelist) package, but not available as such for Pi.
Pi seems to have it in https://github.com/RPi-Distro/firmware-nonfree, but unsure in which repo it resides

macmpi, 2017-06-27:
@heis2201  can you check & confirm if Hotspot works with that dongle/driver?

heis2201, 2017-06-28:
Can't get the hotspot function to work. 

How can I turn the hotspot on (and Wifi client off) in the config files?  The webui doesn't provide of feedback, so I am left dark as to what's not working here.
Thanks.

macmpi, 2017-06-28:
Was expected hotspot won't work actually, driver does not seem to support it: thanks for confirming.

Now, you are on Pi, right?
Since you made several changes back & forth on the firmware install, can you double-check from a clean SD freshly installed (not downloading custom firmware), if it does not work?

As you pointed-out, firmware is now there in initial Volumio Pi image (`/lib/firmware/mt7601u.bin  45 698 bytes`): it is there in my image, and file has same MD5 than the one you downloaded.
So Pi should be covered already, just other Debian may not have it yet.

heis2201, 2017-06-28:
In the meantime, I had tried the [driver](https://github.com/RPi-Distro/firmware-nonfree/blob/master/ralink/mt7601u.bin) that you were pointing at, with which wifi doesn't work at all. (File is also older and has `32 460 bytes`.)

I can try a fresh install on a different SD, but right now I don't have the time, maybe next week. I'll let you know.

Just a question, can the webui handle two wifi interfaces? (i'm thinking for one as wifi client, one as AP)

macmpi, 2017-06-28:
It's a firmware file, not driver: but you probably had download issue, as the file IS the same (and size is 45 412 bytes).

No UI only handles 1 single wifi interface: the primary wlan0.

My mistake, still not in base image; so no need to try yet: but will add it

heis2201, 2017-06-28:
Sure firmware, not driver. It's works as well (when correctly downloaded).

macmpi, 2017-06-28:
added into https://github.com/volumio/Build/pull/213 for Pi

heis2201, 2017-12-28:
is this working for other users on the current release? I still have no luck with this USB dongle on ver 2.348

macmpi, 2017-12-29:
How did you update from previous? Did you try with a fresh install?
I double-checked 2.348 image does contain `/lib/firmware/mt7601u.bin` file as expected, so problem may be different.
Can you check file existence, and report what `dmesg` gives?

closed 2018-03-26 by volumio

jeffreyantony, 2018-04-01:
Hi,

I still can't get the dongle working. Shall I open a new issue or continue over here ?

I tried the following steps
wget -O /lib/firmware/mt7601u.bin git.kernel.org/cgit/linux/kernel/git/firmware/linux-firmware.git/plain/mt7601u.bin

After this I did restart, but still no connection.

volumio@volumio:~$ uname -a
Linux volumio 4.14.29+ #1101 Thu Mar 22 17:06:39 GMT 2018 armv6l GNU/Linux

volumio@volumio:~$ lsusb
Bus 001 Device 051: ID 148f:7601 Ralink Technology, Corp. 
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9512 Standard Microsystems Corp. LAN9500 Ethernet 10/100 Adapter / SMSC9512/9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

DMESG
  683.003331] mt7601u: probe of 1-1.2:1.0 failed with error -110
[  683.011740] usb 1-1.2: USB disconnect, device number 49
[  683.313180] usb 1-1.2: new high-speed USB device number 50 using dwc_otg
[  683.455163] usb 1-1.2: New USB device found, idVendor=148f, idProduct=7601
[  683.455190] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  683.455202] usb 1-1.2: Product: 802.11 n WLAN
[  683.455212] usb 1-1.2: Manufacturer: MediaTek
[  683.455222] usb 1-1.2: SerialNumber: 1.0
[  683.563178] usb 1-1.2: reset high-speed USB device number 50 using dwc_otg
[  683.699192] mt7601u 1-1.2:1.0: ASIC revision: 76010001 MAC revision: 76010500
[  683.701510] mt7601u 1-1.2:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
[  684.156113] mt7601u 1-1.2:1.0: EEPROM ver:0c fae:00
[  684.353780] mt7601u 1-1.2:1.0: Error: RX urb failed:-71
[  684.353899] mt7601u 1-1.2:1.0: Error: RX urb failed:-71

macmpi, 2018-04-02:
Was this dongle working ok on previous Volumio versions like 2.387 (pre 4.14 kernel)?
Could be an issue with `dwc_otg` on 4.14.x (could be improved from 4.14.31).

To confirm, is it working on Raspbian (base 4.9.80 and after `rpi-update` to get to 4.14.31)?

jeffreyantony, 2018-04-02:
I am trying Volumio for the first time. But I had an image from some where beginning of March 2018 in which WiFi didn't work. The current image I use is volumio-2.389-2018-03-26-pi.img.zip which also doesn't work.

I also installed Raspbian and ran the update which brought the kernel to 4.14.31. Here also no success. I get the same error in DMESG output.

macmpi, 2018-04-02:
Didi you also try original Raspbian (4.9.80) before updating?
Kernel's `mt7601u` driver may be faulty for some time...I'll check if there are some replacements available

jeffreyantony, 2018-04-02:
Yes. Both time it didn't work.

macmpi, 2018-04-02:
both are needed: .bin is firmware, .ko is driver.

jeffreyantony, 2018-04-02:
I am not able to install "linux-headers". Any idea what to do ?

macmpi, 2018-04-02:
You may want to refer to [this](https://github.com/kuba-moo/mt7601u/issues/64#issuecomment-324894792): it seems some cheapo dongles need some hack in driver.

jeffreyantony, 2018-04-02:
The problem is this folder
/lib/modules/$(uname -r)/build

I cannot get this. When I try to install the linux headers, I get this
_volumio@volumio:~/mt7601u$ sudo apt-get install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-4.14.29_

jeffreyantony, 2018-04-02:
Hi,

I found in the link you provided, one person mentioned that he solved by using a powered USB hub. I did the same and it worked for me.

Thanks a lot for your quick response and your support.

macmpi, 2018-04-02:
Indeed, I was about to ask if you had good power-supply...
BTW often, using USB hubs with such dongles introduce instabilities: best is direct US connection, with good power supply.
Glad it worked finally.

meudayhegde, 2018-04-12:
Even i have the same issue as @jeffreyantony 
I couldnt install linux-headers for my kernel version-4.14.30-v7+

I installed headers of old kernel
I followed discussion of above issue

I got these errors while make

myBSNL:
make -C /lib/modules/$(uname -r)/build M=$(pwd) modules
make: Entering directory '/lib/modules/4.14.30-v7+/build'
make: *** No rule to make target 'modules'.  Stop.
make: Leaving directory '/lib/modules/4.14.30-v7+/build'

macmpi, 2018-04-12:
Different matter than original title: please submit a new issue.
Original issue seems solved: driver works; it just needs sufficient power-supply (or eventually may work with powered USB hub).

Mridul1129261, 2020-07-06:
Please help me out. I have run dmesg command and found the following list of errors

[   13.817332] mt7601u 2-1.3:1.0: ASIC revision: 76010001 MAC revision: 76010500
[   13.818760] mt7601u 2-1.3:1.0: firmware: failed to load mt7601u.bin (-2)
[   13.818764] mt7601u 2-1.3:1.0: Direct firmware load for mt7601u.bin failed with error -2
[   13.819499] mt7601u: probe of 2-1.3:1.0 failed with error -2
 
Please instruct me what to do next