volumio/Volumio2 #1082: REGDOMAIN for WiFi not set

BorisHollas, 2017-03-20:
The REGDOMAIN in /etc/default/crda is not set, thus WiFi channels 12 & 13 can't be used in the EU.
The user should be able to configure this in the settings.

macmpi, 2017-03-21:
Indeed, is a missing bit following this https://github.com/volumio/Volumio2/issues/327

Would probably make sense to group all locale-related stuff in UI: language (exists), CRDA, [Timezone](https://github.com/volumio/Volumio2/issues/741#issuecomment-286689237), [keyboard](https://volumio.org/forum/installing-the-touch-display-plugin-t6032-20.html#p29849) (issue for chromium-based plugins for instance), [system's locales](https://github.com/volumio/Build/pull/97),...

volumio, 2017-03-21:
Yes, it would make a lot of sense to add all those settings in a unique page. It will take sometime tough...

closed 2017-06-02 by volumio

biva, 2017-06-05:
Hello, does it mean that the issue is solved? Is there a new setting page? Thanks!

BorisHollas, 2017-10-18:
In the current version, it is still not set in  /etc/default/crda.

RD07a, 2017-11-07:
As the language config is already implemented as a dynamic UI setting and the keyboard topic is rather irrelevant for a normal WebUI user I would recommend to consider implementing the rather important "one time" settings of the [WiFi CRDA](https://github.com/volumio/Volumio2/issues/1082) and [timezone ](https://github.com/volumio/Volumio2/issues/741#issuecomment-256066520) into the **First Config Wizard**.

Especially the WiFi CRDA problem could completely block the usage of Volumio via WiFi for some users as their home WiFi may not appear in the available networks list.

volumio, 2017-11-08:
Is there a list of CRDA options that we can provide to the user as selection?

macmpi, 2017-11-08:
info & pointers referred here https://github.com/volumio/Volumio2-UI/issues/317