volumio/Volumio2 #1087: 100% CPU and crash

honza, 2017-03-24:
Today, out of the blue, I started to have this issue. This is on Raspberry Pi.

The server boots, and I access the UI in the browser as usual.

I add a song to the queue or select a song to be played, and the Web UI immediately shows a loading screen because it lost its connection to the server. In the JS console, the socket.io connection is now refused.

When I log into the server via ssh and watch this happen, it appears that the nodejs process quickly climbs to 100% CPU, then systemd determines that it needs to be killed. For a quick moment, there are two nodejs processes, and then systemd kills the rogue one (the first one). Then, I can reload the Web UI again. Except nothing happened. No music is played, nothing was added the queue.

xipmix, 2017-03-24:
Weird. Could you send a log in ? 
 * boot
 * make it crash as described
 * go to /dev url, press 'send report' button.

This should contain everything since the last boot.

honza, 2017-03-24:
@xipmix I don't see a button called "Send report" on that page.

volumio, 2017-03-24:
On which version are you?

honza, 2017-03-24:
2.041 --- Is there an easy way to upgrade?

xipmix, 2017-03-25:
On the <cog menu>->'system' page you should be able to do that.
You may have to visit /dev and tick a box there first;
sorry to be so vague but I can't check easily right now.

Note there is a known issue with upgrading from 2.041
that may break ssh access.
It might be simpler to download a fresh image and set up again.
What things are you concerned about losing if you do that?
The reason I ask is that some work is going on to help export/import
of user data; having some real world user stories would help that.

--

honza, 2017-03-25:
I can confirm that the latest release doesn't suffer from this issue. Thanks for your help!

closed 2017-03-25 by honza