volumio/Volumio2 #1088: Clean Install 2.118 does not work w/o keyboard

mstrangmann, 2017-03-24:
Hello,

on my Rapsi 1 i was not able to make a proper update from 2.114 to 2.118. It said update successful but nothing happens. Perhaps because of the German Language GUI Bug. (i read this later)

So i tried to make a clean install with all the pain of loosing all config. I inserted the SD card in my headless Raspi1 , booted and waited (>10min.). Nothing happened. I retried this several times. Connected to the LAN or just with a Wifi dongle. But i was not able to connect. On the LAN the device had no IP and there was no open wifi-hotspot.

In the end I got a display and a keyboard attached and immediately there was a Volumio hotspot. I don't think this was a coincidence.

Perhaps someone else can verify this behavior.

Kind Regards,
Moritz

xipmix, 2017-03-25:
Could you send in your logs?
 * shutdown & unplug screen & keyboard
 * power up, wait 10 min, attach screen & keyboard as you describe
 * go to /dev url in volumio, press 'send report' button towards the bottom of the page.
This should send a complete log from the last boot up.

mstrangmann, 2017-03-27:
I have no logs, because this just occurs on the very first boot. After i plugged in display and keyboard everything works fine. Also without keyboard.
I will have a look if i can find a second SD-Card to provide the logs.

klokslagB, 2017-03-27:
I also have an issue on my RP2 can't update to 2.118 and also not 2.129. Update says it is successful but after reboot system info still stays 2.114.

klokslagB, 2017-03-27:
Found that deleting the user data solved the issue.