volumio/Volumio2 #1091: No update librairy with nas

Philippe-M, 2017-03-24:
Hello,

After add a network storage (cifs on openmediavault)
[image]

After clicking Update or rescan the library does not update itself even after waiting one night.
[image]

In Browse page i've Musique dir but it's empty
[image]

But with ssh i've access to cifs mount
[image]

volumio, 2017-03-24:
Jus click update database ;)

closed 2017-03-24 by volumio

Philippe-M, 2017-03-24:
Where ?
On Browse page ?

After clic on "Update" bouton i've this in log

017-03-24T16:38:38.105Z - info: [1490373518099] parsing response...
2017-03-24T16:38:38.113Z - info: [1490373518107] parsing response...
2017-03-24T16:38:38.121Z - info: [1490373518115] ControllerMpd::parseState
2017-03-24T16:38:38.173Z - info: Command Router : Notfying DB Updatefalse
2017-03-24T16:38:38.189Z - info: [1490373518183] ControllerMpd::parseState
2017-03-24T16:38:38.197Z - info: [1490373518191] ControllerMpd::pushState
2017-03-24T16:38:38.203Z - info: CoreCommandRouter::servicePushState
2017-03-24T16:38:38.210Z - info: [1490373518204] CoreStateMachine::syncState
2017-03-24T16:38:38.217Z - info: [1490373518211] CorePlayQueue::getTrack 0
2017-03-24T16:38:38.223Z - info: STATE SERVICE {"status":"stop","position":null,"seek":null,"duration":null,"samplerate":null,"bitdepth":null,"channels":null,"random":false,"updatedb":false,"repeat":false,"isStreaming":false,"title":null,"artist":null,"album":null,"uri":null}
2017-03-24T16:38:38.229Z - info: CURRENT POSITION 0
2017-03-24T16:38:38.232Z - info: [1490373518230] CoreStateMachine::syncState   stateService stop
2017-03-24T16:38:38.235Z - info: [1490373518233] CoreStateMachine::syncState   currentStatus stop
2017-03-24T16:38:38.246Z - info: [1490373518236] CoreStateMachine::pushState
2017-03-24T16:38:38.249Z - info: [1490373518246] CoreStateMachine::getState
2017-03-24T16:38:38.251Z - info: [1490373518249] CorePlayQueue::getTrack 0
2017-03-24T16:38:38.254Z - info: CoreCommandRouter::volumioPushState
2017-03-24T16:38:38.257Z - info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
2017-03-24T16:38:38.283Z - info: [1490373518278] interfaceApi::pushState
2017-03-24T16:38:38.290Z - info: [1490373518284] InterfaceWebUI::pushState
2017-03-24T16:38:38.360Z - info: [1490373518354] No code
2017-03-24T16:38:38.367Z - info: [1490373518361] CoreStateMachine::pushState
2017-03-24T16:38:38.373Z - info: [1490373518368] CoreStateMachine::getState
2017-03-24T16:38:38.380Z - info: [1490373518374] CorePlayQueue::getTrack 0
2017-03-24T16:38:38.386Z - info: CoreCommandRouter::volumioPushState
2017-03-24T16:38:38.393Z - info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
2017-03-24T16:38:38.403Z - info: [1490373518397] interfaceApi::pushState
2017-03-24T16:38:38.410Z - info: [1490373518404] InterfaceWebUI::pushState
2017-03-24T16:38:38.488Z - info: [1490373518482] ------------------------------ 855ms