volumio/Volumio2 #1099: On the UI the MPD status about a USB ogg file says 44.1kHz NaN bits

marco-o, 2017-03-27:
The problem is due to a file app/plugins/music_service/mpd/index.js which dues not decodes status text like 44100:f:2. It blindly tries to convert f into a number while f stays for 32bit floating point. Here you get the diff, or I could try to file a pull request.

@@ -517,8 +517,9 @@ ControllerMpd.prototype.parseState = function (objState) {
                var objMetrics = objState.audio.split(':');
                var nSampleRateRaw = Number(objMetrics[0]) / 1000;
                nBitDepth = Number(objMetrics[1])+' bit';
-
-               if (objMetrics[1] == 'dsd') {
+               if (objMetrics[1] == 'f')
+                       nBitDepth = '32f bit';
+               else if (objMetrics[1] == 'dsd') {
                        if (nSampleRateRaw === 352.8) {
                                var nSampleRateRaw = 2.82 + ' MHz';
                                nBitDepth = '1 bit'

volumio, 2017-03-27:
Thanks, please send a PR if possible

xipmix, 2017-06-30:
Hi is this still a problem or can it be closed?