volumio/Volumio2 #1101: Muffled sound after updating to 2.129

mooop12, 2017-03-28:
The sound on S/PDIF output of my HiFiBerry DIGI+, as well as the Pis audio jack output is muffled after updating Volumio to 2.129. I tried reinstalling Volumio on the sd card with no result.

On a side note, the audio output options are finicky to say the least. Sometimes it shows it's outputting to my HiFiBerry when in reality it's outputting to audio jack. When I2S is disabled and you enable it, the DAC list doesn't show up and you can't set your DAC, you have to save it, refresh the page and then change your DAC and save again.

mooop12, 2017-04-08:
Not sure why but going back to 2.118 doesn't fix the issue. The issue is only with Web Radio function, playing regular music files works fine. It surely is a bug in Volumio, because I tried the same stream on Raspbian (using mplayer) and it worked fine.

xipmix, 2017-04-10:
Can you post an example web radio that exhibits the problem? Are there any web radios that work correctly?

mooop12, 2017-04-10:
Sorry, I probably should've checked the earlier... Most of the radios from the Top 500 list actually work fine. The ones that don't work for me I added myself, for example: http://stream.open.fm/94 http://stream.open.fm/14 http://stream.open.fm/3
They use AAC codec if that's relevant.

michalwu, 2017-05-10:
Same here. Open FM sound level is way too low, distorted. 
Is there a place we can enable debug to have logs of what's going on?

xipmix, 2017-05-11:
Go to ```http://<volumioip>/dev/``` and click the 'send logs' link, then paste the log url here. This puts your log on the internetz but it is (lightly) scrubbed beforehand.
Otherwise, login with ssh and run
```sudo journalctl -f```

michalwu, 2017-05-11:
http://logs.volumio.org/volumio/25D9o6d.html
xipmix, thank you.

michalwu, 2017-05-12:
Open.fm is using AAC+ (HE-AAC v2) @ 48 kbps. Is Volumio supporting he-aac v2 profile?

closed 2017-06-02 by volumio

mooop12, 2017-06-03:
@volumio Why was this closed when the issue is not fixed?

michalwu, 2017-06-03:
@volumio Explanation, please?

mooop12, 2017-06-24:
A shame really, it looked like a promising project. I guess open issue statistics are more important than actually fixing bugs, there were over 200 open issues not so long ago, now there are less than 100. A lot of them were closed just like this one with zero response, but I guess it's your project so you're free to drive it into the wall.