volumio/Volumio2 #1102: [2.118] Can not update to 2.129

THEJAJO, 2017-03-28:
I'm not able to update the Volumio 2 (on R Pi 2) to newer version.
Every time I hit Update and restart the Volumio it stays at 2.118.

Is there a way to update via console and don't lose the user data?

volumio, 2017-03-28:
Did you install stuff via apt-get ? Edit manually /etc/os-release? Enabled TEST mode?

bablat, 2017-03-28:
I had the same issue, did not install anything via apt-get. I had to reset all user settings and then upgrade worked.

THEJAJO, 2017-03-28:
I also did not log on to Volumio console and change anything.

Is there a way to backup the user preferences before resetting and upgrading?

THEJAJO, 2017-03-28:
I cleared the data and the Volumio was updated, however old issues came back - Pause button when playing web radio, can not add more than 1 radio from pls playlist (only the first is added, I already submitted the issue few months ago) and probably more.

GetMike, 2017-04-04:
Same problem for me: could not upgrade to 2.129, it even was no longer possible to connect via SSH. After deleting user data the upgrade and also SSH connection worked again.
But now I cannot get any sound out of my hifiberry dac - but it worked with the previous version 2.118.

THEJAJO, 2017-05-12:
Few minutes ago I updated the V2 to 2.163 version without any issue.

closed 2017-06-02 by volumio