volumio/Volumio2 #1111: Refresh page after new plugin install

balbuze, 2017-03-30:
A refresh is required so that the new installed plugin appears. Maybe a "auto-refresh" could be implemented ?

macmpi, 2017-03-30:
Noticed it too, but isn't this a Volumio-UI front-end topic rather?

balbuze, 2017-03-30:
Don't know...But plugin manager is here  : https://github.com/volumio/Volumio2/blob/master/app/pluginmanager.js

xipmix, 2017-03-31:
When the upload finishes one still has the 'upload plugin' tab selected. My impression is that doing wbat balbuze wants means somehow switching to the 'installed plugins' tab. If we know what event is generated by doing that we could perhaps arrange for the end of ```installPlugin``` to ```emiit``` that same event. It would have to occur when the user clicks on 'close' after the upload completes. The 'close' button has an action associated, so perhaps another action could be tacked on, or a more complex action (close and switch to installed plugins tab) could replace the current one.

volumio, 2017-06-02:
https://github.com/volumio/Volumio2/commit/5bfcb44f66b32a8b0cbe8a92662a1f4bf85a2989

closed 2017-06-02 by volumio