volumio/Volumio2 #1114: Plugin manager: improve load order priority management

macmpi, 2017-03-31:
Plugins may have a set boot priority order number in `package.json`.
They are then loaded by such boot priority order by [that code](https://github.com/volumio/Volumio2/blob/master/app/pluginmanager.js#L188-L192): loading is essentially about invoking each `pluginInstance.onVolumioStart()` initialization routine.

However current implementation does NOT guaranties plugin(s) with **boot priority 2** will be initialized **after** all plugin(s) with **boot priority 1** initialization is **completed**.
Therefore, this may lead to unpredictable results if a **boot priority 2** plugin depends on higher priority **boot priority 1** plugin initialization being **completed**.
I took examples of 2 adjacent priorities here, but this could actually span over several priority numbers.

It is expected that **onVolumioStart()** should not take too much time to complete (good practice recommendation probably), but nothing can guaranty actual timing: it is at plugin developer's discretion to ensure.

Therefore, it would be very valuable if `pluginmanager` could ensure that plugins **do complete their initialization by boot priority number batches**, at least: all the # 1 start & complete, then all # 2 start & complete, then...

I'm not familiar enough with promises programming to deals with such type of launch queues concept, but I hope this will be considered.

Thanks for feedback.

macmpi, 2017-04-17:
While trying to manage promises chains in plugin load (onVolumioStart), it seems some plugins have issues:
- 1 _appereance_ and _myvolumio_ plugins miss `return libQ.resolve();` in `onVolumioStart()`
- 2 acces to `PlayerName` seem to cause problems to _upnp_ & _volumio_discovery_ plugins
- 3 volumecontrol.js fails line 37 (if ((cards[device] != undefined) && (cards[device].name != undefined))) 
-...

first item is easy to fix, unsure about others...

macmpi, 2017-04-18:
@volumio would you have some insight on following log, which looks a bit unexpected:

```
Apr 18 16:34:25 volumio volumio[1116]: info: Loading plugin "system"...
Apr 18 16:34:25 volumio volumio[1116]: info: Loading plugin "appearance"...
Apr 18 16:34:27 volumio volumio[1116]: info: Loading plugin "network"...
[...]
Apr 18 16:34:39 volumio volumio[1116]: info: Loading plugin "rest_api"...
Apr 18 16:34:39 volumio volumio[1116]: info: Loading plugin "websocket"...
Apr 18 16:34:39 volumio volumio[1116]: info: Plugin volspotconnect2 is not enabled
Apr 18 16:34:39 volumio volumio[1116]: info: ___________ START PLUGINS ___________
Apr 18 16:34:39 volumio volumio[1116]: PLUGIN START: appearance
Apr 18 16:34:39 volumio volumio[1116]: PLUGIN START: last_100
[...]
Apr 18 16:34:40 volumio volumio[1116]: info: BOOT COMPLETED
[...]
Apr 18 16:34:41 volumio volumio[1116]: Volumio Calling Home
[...]
Apr 18 16:34:44 volumio sudo[1173]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/sbin/i2cdetect -y 1
[...]
Apr 18 16:34:44 volumio volumio[1116]: info: Setting Device type: Raspberry PI
```
Last 3 log entries pertain to `onVolumiostart()` of `system` and `i2s_dac` plugins.
Aren't plugin loads (`onVolumiostart()`) supposed to be all finished before `___________ START PLUGINS ___________` and everything finished before `BOOT COMPLETED`?

Should  `Volumio2/app/index.js` better chain the various actions?
Appreciate your views on this.

macmpi, 2017-05-03:
https://github.com/volumio/Volumio2/pull/1162 intends to improve load/start/stop infrastructure.
**Boot priority number batches** may be implemented after that PR is accepted.
Scheduling issues (from previous post) need to be addressed separately.

fideleinratio, 2017-11-07:
Would like to contribute a solution for this. First, though, need to clarify the scope and requirement : 

Please confirm: am assuming that the plugins should only be started after they have finished loading.

Question 1. What should boot priority apply to: 

- onVolumioStart() ONLY?  (that is, we care what order the plugins finish **Loading**, but after all loaded we do not care what order they finish starting)
- onVolumioStart() then onStart()? (that is we also care that the priority 1 plugins finish **Starting** before starting the priority2's, etc)

Question2: would there be a need to complete the priority1 onVolumioStart() and then onStart() before moving onto priority 2 plugins?  (that is, **Load AND Start all plugins by priority** before moving on to the next plugins)

@macmpi, @fanciulli , @volumio , thanks in advance for your comments.

fideleinratio, 2017-11-07:
And are there any other cases where the order of plugin changes could be important? For example should we execute onVolumioShutdown and onVolumioReboot in the reverse priority order?

fideleinratio, 2017-11-08:
@macmpi  have started looking at this. 

In stepping through there are some parts of Volumio startup in app/index.js that are dependant on completion of at least the higher priority plugin loads. 

Also, there are some plugin onStart functions that are dependant on those non-plugin parts of Volumio startup.

So can mostly ignore my previous comments and questions. The changes are grouped here: 
[https://github.com/fideleinratio/Volumio2/pull/1](url) , which :

- correct some errors in existing plugin promise return from onStart();
- batching by priority for each of plugin.onVolumioStart and plugin.onStart;
- wait for batch completion to continue to next priority;
- further startup actions wait till after plugin load only.  

@macmpi , @fanciulli , @volumio please let me know if any comments or issues.