volumio/Volumio2 #1115: Searching in 'top 500' web radios returns no hits

xipmix, 2017-04-03:
Version 2.129, rpi.

Reproduce: Go to the Browse section, select radio/top500. Type 'Radio' in the search box (my intention was to reduce the number of stations displayed).

Logs:
```
Apr 03 22:41:50 volumio volumio[1425]: info: CoreCommandRouter::volumioGetBrowseSources
Apr 03 22:41:50 volumio volumio[1425]: info: CoreCommandRouter::volumioGetQueue
Apr 03 22:41:50 volumio volumio[1425]: info: [1491259310801] CoreStateMachine::getQueue
Apr 03 22:41:50 volumio volumio[1425]: info: [1491259310804] CorePlayQueue::getQueue
Apr 03 22:41:50 volumio volumio[1425]: info:
Apr 03 22:41:50 volumio volumio[1425]: [1491259310806] ---------------------------- Client requests Volumio queue
Apr 03 22:41:50 volumio volumio[1425]: info: [1491259310814] InterfaceWebUI::pushQueue
Apr 03 22:41:50 volumio volumio[1425]: info: [1491259310823] ------------------------------ 28ms
Apr 03 22:41:52 volumio volumio[1425]: info: CoreCommandRouter::executeOnPlugin: webradio , handleBrowseUri
Apr 03 22:41:52 volumio volumio[1425]: radio
Apr 03 22:41:55 volumio volumio[1425]: info: CoreCommandRouter::executeOnPlugin: webradio , handleBrowseUri
Apr 03 22:41:55 volumio volumio[1425]: radio/top500
Apr 03 22:43:30 volumio volumio[1425]: info: CoreCommandRouter::executeOnPlugin:  , search
Apr 03 22:43:30 volumio volumio[1425]: info: CoreCommandRouter::executeOnPlugin: mpd , search
Apr 03 22:43:30 volumio volumio[1425]: info: CoreCommandRouter::executeOnPlugin: last_100 , search
Apr 03 22:43:30 volumio volumio[1425]: info: Error : CoreCommandRouter::executeOnPlugin: No method [search] in plugin last_100
Apr 03 22:43:30 volumio volumio[1425]: info: CoreCommandRouter::executeOnPlugin: webradio , search
Apr 03 22:43:31 volumio volumio[1425]: info: An error occurred while querying SHOUTCAST

```

liberatorcnx, 2017-04-11:
I think the search function is totally broken.  There is no section in the appropriate file to search for SHOUTCAST type matches. See my issue [here](https://github.com/volumio/Volumio2/issues/1132).  If we can get it working on the library then we can probably do the same for this...

volumio, 2017-06-02:
Fixed with https://github.com/volumio/Volumio2/commit/0584c6b9ecfa29f9c37f83921ad4a91386d00abe

closed 2017-06-02 by volumio