volumio/Volumio2 #1117: no "position" info sent in "pushState" websocket received after getState sent for WebRadio

digx77, 2017-04-05:
If a WebRadio is sent to playing or is currently playing the "getState" command receive a "pushState" reply with MISSED "position" info.
The position info is the info about the position inside the Queue.
In case of normal song playing the position is correctly present in "pushState" reply after the info of "status".

Could you please fix this?

Thanks,
digx

xipmix, 2017-04-10:
I spent some time looking at this but it's not easy to find. Can you explain how you determined this was happening - from the logs? If so, sould you post a log fragment?

digx77, 2017-04-11:
Hi xipmix,
since I develop sound@home I need to parse the websocket messages from Volumio2 and checking the websocket message reply to the "getState" command I receive the "pushState with missed "position" info.
I checked it from my application log, I paste here the messages:

Song:
emit --> ["getState",null]
receiving event --> ["pushState",{"status":"pause","position":1,"title":"Rocco Hunt - Le due storie [prod.Bassi M","artist":null,"album":null,"albumart":"/albumart?path=%2FUSB%2FUSB_01%2FMusic%2FRocco%20Hunt%C3%A9","uri":"mnt/USB/USB_01/Music/Rocco Hunté/Rocco Hunt - Le due storie [prod.Bassi M.mp3","trackType":"mp3","seek":65626,"duration":175,"samplerate":"44.1 KHz","bitdepth":"24 bit","channels":2,"random":null,"repeat":null,"consume":false,"volume":0,"mute":true,"stream":"mp3","updatedb":false,"volatile":false,"service":"mpd"}]

WebRadio:
emit --> ["getState",null]
receiving event --> ["pushState",{"status":"play","title":"<songInfo><show><ScheduleId>74949</ScheduleId><prg_id>403</prg_id><sch_time_from>1500</sch_time_from><sch_time_to>1700</sch_time_to><prg_title>The Flight (lun-ven)</prg_title><image60>http://img.rtl.it/RTLFM/speakers/60x60/la-zac-wxbqp.jpg</image60><image170>http://img.rtl.it/RTLFM/speakers/170x170/la-zac-wxbqp.jpg</image170><image180>http://img.rtl.it/RTLFM/speakers/180x180/la-zac-wxbqp.jpg</image180><image300>http://img.rtl.it/RTLFM/speakers/300x300/fabrizio-ferrari-e7xtd.jpg</image300><image400>http://img.rtl.it/RTLFM/speakers/400x400/fabrizio-ferrari-e7xtd.jpg</image400><speakers>Fabrizio Ferrari, La Zac</speakers><ChaId>1</ChaId></show></songInfo>","artist":"","album":null,"albumart":"/albumart","uri":"http://shoutcast.rtl.it:3010/stream/1/","trackType":"it:3010/stream/1/","seek":129097,"duration":0,"samplerate":"44.1 KHz","bitdepth":"24 bit","channels":2,"random":false,"repeat":false,"repeatSingle":false,"consume":false,"volume":68,"mute":false,"stream":false,"updatedb":false,"volatile":true,"service":"mpd"}]

xipmix, 2017-04-12:
Thanks for the extra info. I have generally been looking at the volumio journal (sudo journalctl -f). I will go have a look at this stream to see if I can find a position item in the state object.

Reformatting your log so I can read it better
```
Song:
emit --> ["getState",null]
receiving event --> [
  "pushState",
  {"status":"pause",
   "position":1,
   "title":"Rocco Hunt - Le due storie [prod.BassiM ",
   "artist":null,
   "album":null,
   "albumart":"/albumart?path=%2FUSB%2FUSB_01%2FMusic%2FRocco%20Hunt%C3%A9",
   "uri":"mnt/USB/USB_01/Music/Rocco Hunté/Rocco Hunt - Le due storie [prod.Bassi M.mp3",
   "trackType":"mp3",
   "seek":65626,
   "duration":175,
   "samplerate":"44.1 KHz",
   "bitdepth":"24 bit",
   "channels":2,
   "random":null,
   "repeat":null,
   "consume":false,
   "volume":0,
   "mute":true,
   "stream":"mp3",
   "updatedb":false,
   "volatile":false,
   "service":"mpd"
   }
 ]

emit --> ["getState",null]
receiving event --> [
  "pushState",
  {"status":"play",
   "title":"74949
             <prg_id>403</prg_id>
             <sch_time_from>1500</sch_time_from>
             <sch_time_to>1700</sch_time_to>
             <prg_title>The Flight (lun-ven)</prg_title>
             http://img.rtl.it/RTLFM/speakers/60x60/la-zac-wxbqp.jpg
             http://img.rtl.it/RTLFM/speakers/170x170/la-zac-wxbqp.jpg
             http://img.rtl.it/RTLFM/speakers/180x180/la-zac-wxbqp.jpg
             http://img.rtl.it/RTLFM/speakers/300x300/fabrizio-ferrari-e7xtd.jpg
             http://img.rtl.it/RTLFM/speakers/400x400/fabrizio-ferrari-e7xtd.jpg
             Fabrizio Ferrari, La Zac1",
   "artist":"",
   "album":null,
   "albumart":"/albumart",
   "uri":"http://shoutcast.rtl.it:3010/stream/1/",
   "trackType":"it:3010/stream/1/",
   "seek":129097,
   "duration":0,
   "samplerate":"44.1 KHz",
   "bitdepth":"24 bit",
   "channels":2,
   "random":false,
   "repeat":false,
   "repeatSingle":false,
   "consume":false,
   "volume":68,
   "mute":false,
   "stream":false,
   "updatedb":false,
   "volatile":true,
   "service":"mpd"
  }
]
```
So yes the position field is indeed missing but most other fields seem to be in order.

The ```title``` attribute is a mess, but testing with vlc suggests that the source is the main problem - I get this mess in the 'now playing' field (reformatted for clarity).
```
<songInfo>
 <show>
  <ScheduleId>75042</ScheduleId>
   <prg_id>361</prg_id>
   <sch_time_from>0900</sch_time_from>
   <sch_time_to>1100</sch_time_to>
   <prg_title>La Famiglia Gi[e][c]249[p] Al Nord</prg_title>
   <image60>http://img.rtl.it/RTLFM/speakers/60x60/silvia-annichiarico-wgcg9.jpg</image60> 
   <image170>http://img.rtl.it/RTLFM/speakers/170x170/fernando-proce-xelgn.jpg</image170>
   <image180>http://img.rtl.it/RTLFM/speakers/180x180/fernando-proce-xelgn.jpg</image180>
   <image300>http://img.rtl.it/RTLFM/speakers/300x300/jennifer-pressman-p3w2x.jpg</image300>
   <image400>http://img.rtl.it/RTLFM/speakers/400x400/silvia-annichiarico-wgcg9.jpg</image400>
   <speakers>Fernando Proce, Jennifer Pressman, Silvia Annichiarico</speakers>
  <ChaId>1</ChaId>
 </show>
 <present>
  <class>Music</class>
  <duration>201684</duration>
  <mus_art_id>589</mus_art_id>
  <mus_art_name>Paola Turci</mus_art_name>
  <mus_art_group>False</mus_art_group>
  <musArtGroupIds />
  <mus_sng_id>35002</mus_sng_id>
  <mus_sng_title>La vita che ho deciso</mus_sng_title>
  <mus_sng_itunesurl>https://itunes.apple.com/it/album/la-vita-che-ho-deciso/id1215779654?i=1215780922[e]amp[p]uo=4[e]amp[p]at=1l3v7yg</mus_sng_itunesurl>
  <mus_sng_itunesreleasedate>31/03/2017</mus_sng_itunesreleasedate>
  <mus_sng_itunesalbumname>Il secondo cuore</mus_sng_itunesalbumname>
  <mus_sng_itunescoverbig>http://is3.mzstatic.com/image/thumb/Music111/v4/9b/ec/98/9bec9832-f13b-f16a-7587-0e7b2543842a/source/600x600bb.jpg</mus_sng_itunescoverbig>
 </present>
</songInfo>
```
The difference in structure shows that ```title``` is getting a bit mangled by something in volumio as well, though.

The ```trackType``` is wrong but this is a known issue, #1116

Have you checked that when a song is in ```"state":"play"``` (your example is showing ```"pause"```) that ```position``` still appears? I guess it does.

xipmix, 2017-04-12:
Have a look at  ```app/statemachine.js CoreStateMachine.prototype.getState```. 

The ```position``` field gets deliberately excluded from what is returned in the 'volatile' and 'consume' cases. I am not sure why.

volumio, 2017-04-12:
It's a bug, position should not be showed in volatile but should be in consume. We'll fix it

digx77, 2017-04-12:
[as I know Consume mode is not supported anymore as per #1119] Are you planning to re-enable it?
However I checked again in the same condition status for song and webradio (no Consume and in play status).
I have same issue:

Song:

-->["pushState",{"status":"play","position":0,"title":"Andiamo a comandare.mp3","artist":"Fabio Rovazzi","album":null,"albumart":"/albumart?web=Fabio%20Rovazzi//extralarge&path=%2FUSB%2FUSB_01%2FMusic","uri":"mnt/USB/USB_01/Music/Andiamo a comandare.mp3","trackType":"mp3","seek":1195,"duration":165,"samplerate":"44.1 KHz","bitdepth":"24 bit","channels":2,"random":null,"repeat":null,"repeatSingle":false,"consume":false,"volume":32,"mute":false,"stream":"mp3","updatedb":false,"volatile":false,"service":"mpd"}]

WebRadio:

-->["pushState",{"status":"play","title":"<songInfo><show><ScheduleId>75043</ScheduleId><prg_id>331</prg_id><sch_time_from>1100</sch_time_from><sch_time_to>1300</sch_time_to><prg_title>W l[e][c]39[p]Italia</prg_title><image60>http://img.rtl.it/RTLFM/team/60x60/angelo-baiguini-valeria-benatti-yeord.jpg</image60><image170>http://img.rtl.it/RTLFM/team/170x170/angelo-baiguini-valeria-benatti-yeord.jpg</image170><image180>http://img.rtl.it/RTLFM/team/180x180/angelo-baiguini-valeria-benatti-yeord.jpg</image180><image300>http://img.rtl.it/RTLFM/team/300x300/angelo-baiguini-valeria-benatti-yeord.jpg</image300><image400>http://img.rtl.it/RTLFM/team/400x400/angelo-baiguini-valeria-benatti-yeord.jpg</image400><speakers>Angelo Baiguini, Valeria Benatti</speakers><ChaId>1</ChaId></show></songInfo>","artist":"","album":null,"albumart":"/albumart","uri":"http://shoutcast.rtl.it:3010/stream/1/","trackType":"it:3010/stream/1/","seek":737,"duration":0,"samplerate":"44.1 KHz","bitdepth":"24 bit","channels":2,"random":false,"repeat":false,"repeatSingle":false,"consume":false,"volume":32,"mute":false,"stream":false,"updatedb":false,"volatile":true,"service":"mpd"}]

P.S.: what is the "volatile" option? Where it can be enabled?

volumio, 2017-04-12:
In fact the consue mode of the statemachine is different:
it allows a service (like webradio) to receive updates from a differet service (like mpd)

the consume mode allows to report status updates from a service with no queue, but as of now nothing in volumio uses it

closed 2017-06-02 by volumio