volumio/Volumio2 #1122: If webradio is playning, next and prev button doesn't work properly

digx77, 2017-04-07:
If webradio isplaying pressing next or prev button the next or prev song in the queue is not played but playing remain on current webradio

ggolec, 2017-04-11:
I'm not quite sure if this happens only by Webradio but I observed the same after couple of days of running volumio. It happens on webradion as well as music from NAS over nfs/smb. When click to next it does nothing. You have to click Next twice then second click does action. It is the same when you have playlist. When click only once nothing happens, you must clich twice to play selected track.

It is important that it happens after few days... Maybe there is some trigger that cause this wierd behavior. 
Just after restart everything works as a charm

Volumio2 on rasp3
vanilla install of 2.129

digx77, 2017-04-11:
IN my case it happens only with WebRadio from the beginning

kregory, 2017-07-21:
only with webradio too for me

re83, 2017-08-27:
I also have this issue

xipmix, 2017-08-30:
I can confirm this on 2.246, rpi2. To reproduce
 * starting from fresh image
 * add one music source (NAS) and let scan complete
 * add one track from one album to the queue
 * add one web radio to the queue
 * add another track from the same album to the queue
 * play first track for ~15s, using 'play' button in 'playback' page
 * press 'next track' - web radio starts to play
 * wait ~15s
 * press 'next track' - nothing happens, nothing is logged
 * press 'previous track' - nothing happens, nothing is logged
 * press 'stop' - web radio stops playing. duration dial is still showing webradio and pulsing.
 * press 'next track' - nothing happens, nothing is logged
 * press 'previous track' - nothing happens, nothing is logged
 * visit 'queue' page and come back to 'playback' page. playback of web radio still stopped. duration dial is still showing webradio and pulsing.
 * press 'next track' - nothing happens, nothing is logged
 * press 'previous track' - nothing happens, nothing is logged

```
2017-08-30T13:06:08.931Z - info: 
[1504098368925] ---------------------------- Client requests Volumio play
2017-08-30T13:06:08.934Z - info: CoreCommandRouter::volumioPlay
2017-08-30T13:06:08.936Z - info: [1504098368935] CoreStateMachine::play index undefined
2017-08-30T13:06:08.937Z - info: [1504098368936] CoreStateMachine::setConsumeUpdateService undefined
2017-08-30T13:06:08.938Z - info: [1504098368937] CorePlayQueue::getTrack 0
2017-08-30T13:06:08.939Z - info: [1504098368939] CoreStateMachine::startPlaybackTimer
2017-08-30T13:06:08.940Z - info: [1504098368940] CorePlayQueue::getTrack 0
2017-08-30T13:06:08.953Z - info: [1504098368952] ControllerMpd::resume
2017-08-30T13:06:08.955Z - info: [1504098368954] ControllerMpd::sendMpdCommand play
2017-08-30T13:06:08.965Z - info: [1504098368962] sending command...
2017-08-30T13:06:08.976Z - info: [1504098368972] ------------------------------ 47ms
2017-08-30T13:06:08.985Z - info: Mpd Status Update: player
2017-08-30T13:06:08.990Z - info: 
[1504098368985] ---------------------------- MPD announces state update
2017-08-30T13:06:08.995Z - info: [1504098368991] parsing response...
2017-08-30T13:06:08.998Z - info: [1504098368995] ControllerMpd::getState
2017-08-30T13:06:09.002Z - info: [1504098368999] ControllerMpd::sendMpdCommand status
2017-08-30T13:06:09.003Z - info: [1504098369002] sending command...
2017-08-30T13:06:09.008Z - info: [1504098369007] parsing response...
2017-08-30T13:06:09.010Z - info: [1504098369009] ControllerMpd::parseState
2017-08-30T13:06:09.011Z - info: [1504098369010] ControllerMpd::sendMpdCommand playlistinfo
2017-08-30T13:06:09.013Z - info: [1504098369012] sending command...
2017-08-30T13:06:09.019Z - info: [1504098369018] parsing response...
2017-08-30T13:06:09.022Z - info: [1504098369021] ControllerMpd::parseTrackInfo
2017-08-30T13:06:09.024Z - info: [1504098369023] ControllerMpd::pushState
2017-08-30T13:06:09.025Z - info: CoreCommandRouter::servicePushState
2017-08-30T13:06:09.026Z - info: [1504098369025] CoreStateMachine::syncState
2017-08-30T13:06:09.027Z - info: [1504098369026] CorePlayQueue::getTrack 0
2017-08-30T13:06:09.028Z - info: STATE SERVICE {"status":"play","position":0,"seek":9450,"duration":179,"samplerate":"44.1 KHz","bitdepth":"16 bit","channels":2,"random":false,"updatedb":false,"repeat":false,"isStreaming":false,"title":"Desire","artist":"U2","album":"Rattle and Hum","uri":"NAS/testu2/Rattle_and_Hum/03-Desire.flac","trackType":"flac"}
2017-08-30T13:06:09.029Z - info: CURRENT POSITION 0
2017-08-30T13:06:09.030Z - info: [1504098369030] CoreStateMachine::syncState   stateService play
2017-08-30T13:06:09.031Z - info: [1504098369031] CoreStateMachine::syncState   currentStatus pause
2017-08-30T13:06:09.033Z - info: [1504098369032] CoreStateMachine::pushState
2017-08-30T13:06:09.034Z - info: [1504098369033] CoreStateMachine::getState
2017-08-30T13:06:09.035Z - info: [1504098369034] CorePlayQueue::getTrack 0
2017-08-30T13:06:09.036Z - info: CoreCommandRouter::volumioPushState
2017-08-30T13:06:09.037Z - info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
2017-08-30T13:06:09.039Z - info: [1504098369038] interfaceApi::pushState
2017-08-30T13:06:09.041Z - info: [1504098369040] InterfaceWebUI::pushState
2017-08-30T13:06:09.082Z - info: [1504098369080] CoreStateMachine::pushState
2017-08-30T13:06:09.083Z - info: [1504098369082] CoreStateMachine::getState
2017-08-30T13:06:09.085Z - info: [1504098369084] CorePlayQueue::getTrack 0
2017-08-30T13:06:09.086Z - info: CoreCommandRouter::volumioPushState
2017-08-30T13:06:09.088Z - info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
2017-08-30T13:06:09.091Z - info: [1504098369090] interfaceApi::pushState
2017-08-30T13:06:09.093Z - info: [1504098369092] InterfaceWebUI::pushState
2017-08-30T13:06:09.142Z - info: [1504098369140] ------------------------------ 160ms
2017-08-30T13:06:09.154Z - info: Pushing Favourites {"service":"mpd","uri":"mnt/NAS/testu2/Rattle_and_Hum/03-Desire.flac","favourite":false}
2017-08-30T13:06:09.161Z - info: Pushing Favourites {"service":"mpd","uri":"mnt/NAS/testu2/Rattle_and_Hum/03-Desire.flac","favourite":false}

[... 322 lines cut ...]
2017-08-30T13:08:40.461Z - info: [1504098520460] ControllerMpd::sendMpdCommand stop
2017-08-30T13:08:40.469Z - info: [1504098520468] sending command...
[... cut ...]
2017-08-30T13:08:40.595Z - info: [1504098520594] ControllerMpd::pushState
2017-08-30T13:08:40.596Z - info: CoreCommandRouter::servicePushState
2017-08-30T13:08:40.597Z - info: [1504098520596] CoreStateMachine::syncState
2017-08-30T13:08:40.598Z - info: [1504098520597] CorePlayQueue::getTrack 1
2017-08-30T13:08:40.599Z - info: [1504098520598] Received update from a service different from the one supposed to be playing music. Skipping notification.Current webradio Received mpd
2017-08-30T13:08:40.600Z - info: [1504098520599] ------------------------------ 25ms

```

xipmix, 2017-08-30:
When I remove the webradio item from the queue (playback still stopped), the playback page moves on to the next album track. I can then press play and move back and forward between the album tracks with no issue.

xipmix, 2017-09-01:
After a bit of poking and prying I found some problems with webradio handling, but not the cause of this issue. Then I turned to the [UI code](https://github.com/volumio/Volumio2-UI/blob/a7697e36b84c00369d400c540809c4ca3d69cdcd/src/app/services/player.service.js#L70)

So this appears to be deliberate ... but evidently not a behaviour users are expecting.

I'm also unsure how ```trackType``` ever gets to be ```webradio```. In the log excerpt above ```trackType``` is actually ```com/ABC-Lounge```.
The original url of the webradio is                                             
```http://yp.shoutcast.com/sbin/tunein-station.m3u?id=709809```                 
The URL that comes back from the service (in a .m3u file) is                    
```http://listen.radionomy.com/ABC-Lounge.com/ABC-Lounge```                     
which is then mangled into these fields:
```uri: "http://listen.radionomy.com/ABC-Lounge"``` and ```trackType: "com/ABC-Lounge"```
by  https://github.com/volumio/Volumio2/blob/bbe4e278112a0ad5d266695f3c6741b016592d16/app/plugins/music_service/mpd/index.js#L378

I can see where the state machine tries to set it to ```webradio```: https://github.com/volumio/Volumio2/blob/bbe4e278112a0ad5d266695f3c6741b016592d16/app/statemachine.js#L104
but it's unclear whether that works in this case or is even relevant.

xipmix, 2017-09-03:
So if I disable the restriction about webradio in the UI next function and then press next while playing a webradio, I get:
```
Sep 03 01:19:21 volumio volumio[3840]: info:                                    
Sep 03 01:19:21 volumio volumio[3840]: [1504401561624] ---------------------------- Client requests Volumio next
Sep 03 01:19:21 volumio volumio[3840]: info: CoreCommandRouter::volumioNext     
Sep 03 01:19:21 volumio volumio[3840]: info: [1504401561634] CoreStateMachine::next
Sep 03 01:19:21 volumio volumio[3840]: info: Error:                             
Sep 03 01:19:21 volumio volumio[3840]: TypeError: thisPlugin.next is not a function
Sep 03 01:19:21 volumio volumio[3840]: at CoreStateMachine.next (/volumio/app/statemachine.js:1162:15)
Sep 03 01:19:21 volumio volumio[3840]: at CoreCommandRouter.volumioNext (/volumio/app/index.js:108:27)
Sep 03 01:19:21 volumio volumio[3840]: at nextTickCallback (/volumio/node_modules/kew/kew.js:47:28)
Sep 03 01:19:21 volumio volumio[3840]: at _combinedTickCallback (internal/process/next_tick.js:73:7)
Sep 03 01:19:21 volumio volumio[3840]: at process._tickCallback (internal/process/next_tick.js:104:9)
Sep 03 01:19:21 volumio volumio[3840]: info: [1504401561649] ------------------------------ 25ms
Sep 03 01:19:40 volumio volumio[3840]: info:                                    
```
With the restriction still disabled, and the webradio track selected but stopped, pressing 'next' does the right thing.
```
<pressed stop>                                                                 
Sep 03 01:19:40 volumio volumio[3840]: info:                                    
Sep 03 01:19:40 volumio volumio[3840]: [1504401580347] ---------------------------- Client requests Volumio stop
Sep 03 01:19:40 volumio volumio[3840]: info: CoreCommandRouter::volumioStop     
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580356] CoreStateMachine::stop
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580358] CoreStateMachine::setConsumeUpdateService undefined
Sep 03 01:19:40 volumio volumio[3840]: UNSET VOLATILE                           
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580361] CoreStateMachine::stPlaybackTimer
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580363] CoreStateMachine::updateTrackBlock
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580367] CorePlayQueue::getTrackBlock
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580370] CoreStateMachine::pushState
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580371] CoreStateMachine::getState
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580373] CorePlayQueue::getTrack 1
Sep 03 01:19:40 volumio volumio[3840]: info: CoreCommandRouter::volumioPushState
Sep 03 01:19:40 volumio volumio[3840]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580381] interfaceApi::pushState
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580382] InterfaceWebUI::pushState
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580432] CoreStateMachine::serviceStop
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580434] CorePlayQueue::getTrack 1
Sep 03 01:19:40 volumio volumio[3840]: info: CoreCommandRouter::volumioPushState
Sep 03 01:19:40 volumio volumio[3840]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580381] interfaceApi::pushState
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580382] InterfaceWebUI::pushState
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580432] CoreStateMachine::serviceStop
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580434] CorePlayQueue::getTrack 1
Sep 03 01:19:40 volumio volumio[3840]: info: CoreCommandRouter::serviceStop     
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580439] ControllerWebradio::stop
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580441] ControllerMpd::sendMpdCommand stop
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580451] sending command... 
Sep 03 01:19:40 volumio volumio[3840]: info: Pushing Favourites {"service":"webradio","uri":"http://yp.shoutcast.com/sbin/tunein-  station.m3u?id=709809","favourite":false}
Sep 03 01:19:40 volumio volumio[3840]: info: No input data                      
Sep 03 01:19:40 volumio volumio[3840]: info: Mpd Status Update: player          
Sep 03 01:19:40 volumio volumio[3840]: info:                                    
Sep 03 01:19:40 volumio volumio[3840]: [1504401580571] ---------------------------- MPD announces state update
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580574] parsing response...
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580576] ControllerMpd::getState
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580578] ControllerMpd::sendMpdCommand status
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580580] ------------------------------ 233ms
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580582] sending command... 
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580587] parsing response...
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580590] ControllerMpd::parseState
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580592] ControllerMpd::sendMpdCommand playlistinfo
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580594] sending command... 
Sep 03 01:19:40 volumio volumio[3840]: info: [1504401580599] parsing response...
[... cut ...]
Sep 03 01:21:47 volumio volumio[3840]: info: Pushing Favourites {"service":"mpd","uri":"mnt/NAS/testu2/Rattle_and_Hum/04-          Hawkmoon_269.flac","favourite":false}
Sep 03 01:21:47 volumio volumio[3840]: info: Pushing Favourites {"service":"mpd","uri":"mnt/NAS/testu2/Rattle_and_Hum/04-          Hawkmoon_269.flac","favourite":false}
Sep 03 01:22:05 volumio volumio[3840]: info:                                    
Sep 03 01:22:05 volumio volumio[3840]: [1504401725310] ---------------------------- Client requests Volumio pause
Sep 03 01:22:05 volumio volumio[3840]: info: CoreCommandRouter::volumioPause    
```

p1234321, 2017-10-25:
hello,
it would really be aweseome to be able to navigate through webradios and make a playlist of webradios!!!

R13e, 2018-01-24:
hey,
any progress? the bug still exists.

TaliesinDS, 2018-01-30:
I would be really nice if this was fixed.
I based a project around this working properly :(

diehardsk, 2018-02-08:
Hi everyone,
I'm in the same opinion. To build an internet radio player for everyday use hardware buttons or rotary encoder control is necessary.
I followed xipmix's analyse and found we can quite easily overcome problem happening in next/prev function. Modified line 1166 and 1306  in statemachine.js
from
`		if (this.isConsume && this.consumeState.service != undefined) {`
to
`		if (this.isConsume && this.consumeState.service != undefined && this.consumeState.service !== 'webradio') {
`

And it works just fine. Anyway, I'm new to Volumio and js, so you'd better consider it as a dirty hotfix. Also to be able to use next/prev/repeat in web ui you need to remove webradio restriction as described above.

serperus, 2018-04-20:
Hi,

I'm very sorry, but how to 

" you need to remove webradio restriction as described above"

Please point me to the right code for that.

Thank you!

diehardsk, 2018-04-20:
@serperus 
if you look at volumio/http/www/scripts/app-697d229700.js
you can find the restriction `"webradio"!==this.state.trackType&&` in some button's function.
Save original file and than you can experiment by removing this code.

serperus, 2018-04-20:
@diehardsk
thank you!
I have found this file with grep "webradio" but it seems to be hard to edit and it not so humanreadable :)

but I'll try.

thanks again

serperus, 2018-07-04:
Hi,

In order to do next&prev buttons working with webradio you should do the following:

1) modify statemachine.js as described below by @diehardsk in both functuions (thisPlugin.next and thisPlugin.previous)
2) modify file /volumio/http/www/scripts/app-<your-version>.js and remove code 

"webradio"!==this.state.trackType&& from this file (for next and for prev)

Please do backups before performing any modification!

Thank you @diehardsk for your help.

Works like a charm :)

pagaille, 2018-07-21:
Working too. 
I'd like to submit a PR, but does it make sense to submit it for the js file ?

xipmix, 2018-07-25:
In the Volumio2-UI project I think this code   https://github.com/volumio/Volumio2-UI/blob/88a6a0c0487ebf5e70da7360ed77673117e6946b/src/app/services/player.service.js#L64-L74 that  is what ends up in http/www/scripts/app-your-version.js. I am not sure  what transformation is applied to get it there, but I think the change needed is:
```
--- a/src/app/services/player.service.js
+++ b/src/app/services/player.service.js
@@ -62,15 +62,11 @@ class PlayerService {
   }
 
   prev() {
-    if (this.state.trackType !== 'webradio') {
       this.socketService.emit('prev');
-    }
   }
 
   next() {
-    if (this.state.trackType !== 'webradio') {
       this.socketService.emit('next');
-    }
   }
 
   shuffle() {

```
So two PRs will be needed to get this in, one for statemachine.js in volumio/Volumio2 and one for player.service.js in volumio/Volumio2-UI. 

The combination of both will need to be tested. To test the staemachine.js change, just upload the modified file to your volumio and reboot. The UI project is a little more tricky, see https://github.com/volumio/docs/blob/master/docs/01_Development_How_To/04_Set_up_development_environment_for_UI.md . Basically you modify the player.service.js file in your git clone, build the project and it produces new stuff in the ```dist/``` subdirectory. Copy the contents of ```dist/*``` dir to ```volumio@volumio:/volumio/http/www/``` and reboot volumio again.
@balbuze you made some updates to that section of the docs recently, can we use rsync to copy things over, ie ```rsync -uv dist/ volumio@volumio:/volumio/http/www/``` ?

ZigUSSS, 2018-12-02:
does anyone do anything about the commands in webradio next prev

tiroy1, 2019-05-21:
How the heck such a simple fix is not already in released code?

TonyVpck, 2020-01-05:
Hello everyone,
An update have been made since the "next/previous webradio" issue was identify but, i've just installed volumio, and the problem remain the same. 
Can I modify JS files with my navigator ? Do you think this issue will be solved for a next update ?
Have a good day !
Tony

TonyVpck, 2020-01-08:
Hello !
I forked the Volumio folders and modified the JS files to be able to test, but I can't find the original JS codes in the tree of my raspberry pi.
Did anyone test functional changes that could be added to a futur pull request?
Have a good day !

xipmix, 2020-01-09:
Changing the UI code is a bit awkward because it gets minified. This page explains: https://volumio.github.io/docs/Development_How_To/Set_up_development_environment_for_UI.html
The trick is the last step - manually copying the ```dist``` directory to ```/volumio/http/www```.
What I've done in the past is
```
$ rsync -rvt dists/ volumo@volumio:/volumio/http/www
```
Can you give that a try and let us know how you get on?

Not sure what you mean by "functional changes" - do you mean changes to the backend code?

TonyVpck, 2020-01-11:
Hello,

I just made the manual modifications of this files from the console : 
statemachine.js and "yourversionnumber".js

The changes seem to make the next and prev buttons work !

Do you think it is necessary to make a pull request to bring a solution to the next versions of Volumio ? Do we have to do other tests to make sure the buttons solution is okay ? (I'm new to github, I'm not sure how we can suggest this to developers and who can suggest it).

Have a good day!

xipmix, 2020-01-13:
Definitely. You'll need to do two PRs, one for Volumio2 (statemachine.js) and one for Volumio2-UI. Possibly it will help if you could post your changes here inside a code block (three ` marks before and after, on a separate line). Then we can figure out how to prepare the PRs.

As I said the app-yourversionnumber_.js file is a minified version compiled from lots of little bits of code in the Volumio2-UI project. We'll have to find the original files and change those for the pull request to be accepted. Should not be that difficult.

TonyVpck, 2020-01-13:
To sum up, the modification should be :

```   
@@ -1267,7 +1267,7 @@ CoreStateMachine.prototype.next = function (promisedResponse) {
        }
	} else {
		//self.setConsumeUpdateService(undefined);
		if (this.isConsume && this.consumeState.service != undefined) {
		if (this.isConsume && this.consumeState.service != undefined && this.consumeState.service !== 'webradio') { 
			var thisPlugin = this.commandRouter.pluginManager.getPlugin('music_service', this.consumeState.service);
            if (typeof thisPlugin.next === "function") {
                thisPlugin.next();
@@ -1427,7 +1427,7 @@ CoreStateMachine.prototype.previous = function (promisedResponse) {
			}

		} else if (this.currentStatus === 'play') {
			if (this.isConsume && this.consumeState.service != undefined) {
			if (this.isConsume && this.consumeState.service != undefined && this.consumeState.service !== 'webradio') { 
                var thisPlugin = this.commandRouter.getMusicPlugin(this.consumeState.service);
                if (!this.previousTrackonPrev && typeof thisPlugin.seek === "function") {
                    thisPlugin.seek(0);
```

and as you were able to find it previously, in player.service.js :

```   
@@ -69,15 +69,11 @@ class PlayerService {
  }

  prev() {
    if (this.state.trackType !== 'webradio') {
      this.socketService.emit('prev');
    }
  }

  next() {
    if (this.state.trackType !== 'webradio') {
      this.socketService.emit('next');
    }
  }

  skipBackwards() {
```

Have a nice evening !

xipmix, 2020-01-14:
Thanks for this. I've been holding off on submitting this for lack of time to figure out how to test the modified UI code, I was stumbling over the minification step. But you've worked that out (by modifying the minifed code - did you beautify it first?), so let's get this done.

First are there changes of behaviour for other sources, like tracks from the music library? And a mix of sources? You want to make sure your change doesn't introduce regressions.

Here's the step by step
 * fork Volumio2 and Volumio2-UI
 * clone those locally
 * in each make a new branch and commit the changes to statemanager.js and player.service.js, respectively
 * push back to your origin

This will trigger a change in the github web UI suggesting you submit a pull request (actually two).
When you do that, you should mention in the comments what platform you tested on and what tests you did. You should mention this issue too.

TonyVpck, 2020-01-15:
Hello !

I just tried to play webradio mixed with flac and mp3 files, everything seems to work fine.

The only thing I noticed that overlaps with a previously written post:
It might be a good idea not to make the change just for the next and prev buttons, but extend this to the shuffle and repeat buttons.

The reason is that it is possible to activate shuffle and repeat only on audio files (and not on webradio) while it works for the entire queue !

I plan to add these new changes to my fork before submitting the PR? What do you think ?

Good evening !
Tony

xipmix, 2020-01-15:
go for it, I reckon it will work

TonyVpck, 2020-01-16:
The PR is okay, the modification have been merged !

Thank you for help !
See you next time !

xipmix, 2020-01-30:
Can someone close this please, it's done.

closed 2020-01-30 by volumio

DaniilMir, 2022-06-05:
For Volumio 3 even with this fix it was not switching to previous stream. Next stream was working properly. 
I found a way to make it work : 
File : /volumio/app/statemachine.js

Replace `this.previousTrackonPrev = false;`   with  `this.previousTrackonPrev = true;`

Not sure if this parameter "previousTrackonPrev" was designed to be configurable from UI. 
But it worked.