volumio/Volumio2 #1123: Can't Install volspotconnect2.zip

psychophil, 2017-04-08:
I'm getting install errors. Tried on pi B, pi 2 and pi 3. With Volumio 2.137 and latest stable release from website. Using your latest zip installation stalls at creating folder. On 1st March zip it stalls at downloading dependency after arm7, arm6 detected. Also tried different SD cards. Also trief downloading​ your zip several times too.
All was well with pre March versions but I've deleted them. Perhaps you can make the earlier zip available.

apinner, 2017-04-08:
Same issue here.

xipmix, 2017-04-10:
Can we haz logs pls. Go to /dev, press 'send log' button.

apinner, 2017-04-11:
http://logs.volumio.org/volumio/CUkAMVA.html

volumio, 2017-04-11:
There are issues with the install script:

Apr 11 06:01:01 volumio volumio[1115]: [sudo] password for volumio: converted 'https://github.com/balbuze/volumio-plugins/raw/master/plugins/music_service/volspotconnect2/arch/armhf/librespot-armhf.tar.xz' (ANSI_X3.4-1968) -> 'https://github.com/balbuze/volumio-plugins/raw/master/plugins/music_service/volspotconnect2/arch/armhf/librespot-armhf.tar.xz' (UTF-8)
Apr 11 06:01:01 volumio volumio[1115]: --2017-04-11 06:00:59--  https://github.com/balbuze/volumio-plugins/raw/master/plugins/music_service/volspotconnect2/arch/armhf/librespot-armhf.tar.xz
Apr 11 06:01:01 volumio volumio[1115]: Resolving github.com (github.com)... 192.30.253.112, 192.30.253.113
Apr 11 06:01:01 volumio volumio[1115]: Connecting to github.com (github.com)|192.30.253.112|:443... connected.
Apr 11 06:01:01 volumio volumio[1115]: HTTP request sent, awaiting response... 302 Found
Apr 11 06:01:01 volumio volumio[1115]: Location: https://raw.githubusercontent.com/balbuze/volumio-plugins/master/plugins/music_service/volspotconnect2/arch/armhf/librespot-armhf.tar.xz [following]
Apr 11 06:01:01 volumio volumio[1115]: converted 'https://raw.githubusercontent.com/balbuze/volumio-plugins/master/plugins/music_service/volspotconnect2/arch/armhf/librespot-armhf.tar.xz' (ANSI_X3.4-1968) -> 'https://raw.githubusercontent.com/balbuze/volumio-plugins/master/plugins/music_service/volspotconnect2/arch/armhf/librespot-armhf.tar.xz' (UTF-8)
Apr 11 06:01:01 volumio volumio[1115]: --2017-04-11 06:01:01--  https://raw.githubusercontent.com/balbuze/volumio-plugins/master/plugins/music_service/volspotconnect2/arch/armhf/librespot-armhf.tar.xz
Apr 11 06:01:01 volumio volumio[1115]: Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 62.252.172.241
Apr 11 06:01:01 volumio volumio[1115]: Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|62.252.172.241|:443... connected.
Apr 11 06:01:01 volumio volumio[1115]: GnuTLS: A TLS fatal alert has been received.
Apr 11 06:01:01 volumio volumio[1115]: GnuTLS: received alert [0]: Close notify
Apr 11 06:01:01 volumio volumio[1115]: Unable to establish SSL connection.
Apr 11 06:01:01 volumio volumio[1115]: /data/plugins//music_service/volspotconnect2/install.sh: 34: exit: Illegal number: -1
Apr 11 06:01:01 volumio volumio[1115]: info: Install script return the error Error: Command failed: echo volumio | sudo -S sh /data/plugins//music_service/volspotconnect2/install.sh > /tmp/installog
Apr 11 06:01:01 volumio volumio[1115]: [sudo] password for volumio: converted 'https://github.com/balbuze/volumio-plugins/raw/master/plugins/music_service/volspotconnect2/arch/armhf/librespot-armhf.tar.xz' (ANSI_X3.4-1968) -> 'https://github.com/balbuze/volumio-plugins/raw/master/plugins/music_service/volspotconnect2/arch/armhf/librespot-armhf.tar.xz' (UTF-8)
Apr 11 06:01:01 volumio volumio[1115]: --2017-04-11 06:00:59--  https://github.com/balbuze/volumio-plugins/raw/master/plugins/music_service/volspotconnect2/arch/armhf/librespot-armhf.tar.xz
Apr 11 06:01:01 volumio volumio[1115]: Resolving github.com (github.com)... 192.30.253.112, 192.30.253.113
Apr 11 06:01:01 volumio volumio[1115]: Connecting to github.com (github.com)|192.30.253.112|:443... connected.
Apr 11 06:01:01 volumio volumio[1115]: HTTP request sent, awaiting response... 302 Found
Apr 11 06:01:01 volumio volumio[1115]: Location: https://raw.githubusercontent.com/balbuze/volumio-plugins/master/plugins/music_service/volspotconnect2/arch/armhf/librespot-armhf.tar.xz [following]
Apr 11 06:01:01 volumio volumio[1115]: converted 'https://raw.githubusercontent.com/balbuze/volumio-plugins/master/plugins/music_service/volspotconnect2/arch/armhf/librespot-armhf.tar.xz' (ANSI_X3.4-1968) -> 'https://raw.githubusercontent.com/balbuze/volumio-plugins/master/plugins/music_service/volspotconnect2/arch/armhf/librespot-armhf.tar.xz' (UTF-8)
Apr 11 06:01:01 volumio volumio[1115]: --2017-04-11 06:01:01--  https://raw.githubusercontent.com/balbuze/volumio-plugins/master/plugins/music_service/volspotconnect2/arch/armhf/librespot-armhf.tar.xz
Apr 11 06:01:01 volumio volumio[1115]: Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 62.252.172.241
Apr 11 06:01:01 volumio volumio[1115]: Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|62.252.172.241|:443... connected.
Apr 11 06:01:01 volumio volumio[1115]: GnuTLS: A TLS fatal alert has been received.
Apr 11 06:01:01 volumio volumio[1115]: GnuTLS: received alert [0]: Close notify
Apr 11 06:01:01 volumio volumio[1115]: Unable to establish SSL connection.
Apr 11 06:01:01 volumio volumio[1115]: /data/plugins//music_service/volspotconnect2/install.sh: 34: exit: Illegal number: -1
Apr 11 06:01:01 volumio volumio[1115]: info: An error occurred installing the plugin. Rolling back config

You should report to the plugin developer here:
https://github.com/balbuze/volumio-plugins/tree/master/plugins/music_service/volspotconnect2

balbuze, 2017-04-14:
I just discovered this. I retest and I found no error in install script. It install with no problem on my RPI2 and PINE64.... And nobody report this problem. There is something else (network issue ?) and install.sh returns an error because it is designed like this, if something is wrong during download it exit installation.
			echo "Failed to download. Stopping installation now"
			exit -1
		fi
Anyway if you still have problem re-contact me https://github.com/balbuze/volumio-plugins/tree/master/plugins/music_service/volspotconnect2 and have a look at issues.
Thanks

jerryrard, 2017-05-15:
Mine installed fine. However when I had my friend install his, he also is getting an error... right after "Creating folder on disk"
"The following error occurred when installing the plugin: Error"
I installed from Windows, and had no problem. My friend is installing from a Mac, and is getting the error. I don't think that the OS should matter.

psychophil, 2017-05-15:
I gave up and brought a Chromecast Audio. However I find I much prefer the pi/digi+/volumio2 sound. So I've retried whenever there is an update to see if it'll work. Sunday I successfully installed the plugin. So decided to make second Install card as a backup but cannot get it to install due to the download error. I can manually get the file on my Linux box so it's NOT the ISP! Virgin Media 100MBs. Just keeping my fingers crossed the working install doesn't fail again. My suspicion is a timeout error at the server end. I tried dozens and dozens of times over a 2 Months before it worked just the once!

xipmix, 2017-05-15:
Can you send logs of your failing downloads (go to ```http://volumio/dev```, press 'send logs')?
If you login to the pi and run 'df -Ph' what does the output show - is e.g. /tmp full?

macmpi, 2017-05-15:
@jerryrard wrote
> I installed from Windows, and had no problem. My friend is installing from a Mac, and is getting the error.

which Volumio version are you using?
Recently (can't remember which which release it came) @xipmix fixed some issues with manual download files names.
If you use older Volumio, make sure uploaded file name do not have uncommon characters like () spaces, etc...

balbuze, 2017-05-15:
@psychophil even if you can download file, have you tried to set a custom dns in volumio ? ( at the bottom of the network section).

psychophil, 2017-05-15:
I did try OpenDNS and Google settings I got from another thread but they
didn't help, but it did work for someone on another thread somewhere.

balbuze, 2017-05-15:
tell me on what  device you want to install. I can send you a modified install.sh to install with local file. Or you can comment (add #) on each line with wget in install.sh and unzip the right librespot download from my repo in plugin folder.

jerryrard, 2017-05-15:
@macmpi 
I am using Ver 2.041, and it is working fine. I just dragged the ZIP file to the Web UI, and it installed without any issues.

balbuze, 2017-05-15:
Special version to install from local files.
You have first of all to copy the correct librespot.YYY.xz file for your device in /data/INTERNAL without decompressing it. https://github.com/balbuze/volumio-plugins/tree/master/plugins/music_service/volspotconnect2/arch
arm = rpi 0, B
armhf = rpi 2 3 pine64 sparky and any armv7 device
x86 = x86 computer

The use the following to install I DIDNT TESTED IT!!!
[volspotconnect2.zip](https://github.com/volumio/Volumio2/files/1002145/volspotconnect2.zip)

balbuze, 2017-05-15:
I was thinking... Have tried to install using a click in upload area and selecting the file with file browser opened? (Instead of drag and drop)

macmpi, 2017-05-15:
@jerryrard 
This version **is sensitive to file names** for sure (it is quite old version).
If your PC/Mac alters the .zip filename using custom characters (like on MacOS if you have several copies of the same file), plugin install will fail with such file.
This is not a volspotconnect specific issue, it is a plugin install procedure issue.
This (and quite a lot of other things) have been improved since 2.041: you're certainly better off upgrading.

jerryrard, 2017-05-16:
@macmpi 
My 2.041 version is working with volspotconnect. I have NO problems.
It is my FRIEND who has a problem installing it and his Volumio version is 2.163.

jerryrard, 2017-05-16:
@macmpi 
You guys are correct! Thanks for all your help.
The MAC may have modified the filename. 
My friend's volspotconnect is UP and RUNNING.
Thank you so very much!

macmpi, 2017-05-16:
@psychophil 

> I did try OpenDNS and Google settings I got from another thread

Which Volumio version are you using?
DNS setup through Volumio UI has been improved/fixed with 2.163

psychophil, 2017-05-16:
Version 2.169
Your plugin is showing as 2.4 on the working install. Have a second pi to
experiment with. Curious I got it to install only once. Last February was
last successful install before Sunday and that took a couple of tries.
Already tried changing the wget line to get file of my WDMyCloud device.
Will try the above suggestions soon-ish.

closed 2017-06-02 by volumio