volumio/Volumio2 #1128: setRandom websocket interface problems...

DevBCN456, 2017-04-10:
Hi all,
I'm working on a program with python 2.7/ socketIO-client 0.7.2 package and Volumio websocket interface. Eerything works properly, I can access and load my Playlists from the program, use local buttons with the GPIO library (play/stop/pause) to control basic playback functions... but I'd need to set the Random property when I load my Playlists from the program.
I have realized this code works perfectly...
       socketIO.emit('setRandom', False)
       socketIO.emit('setRandom', 0)
but this one doesn't.....
       socketIO.emit('setRandom' True)
       socketIO.emit('setRandom', 1)
Any idea or could someone suggests a workaround to set the random property by default when loading the Playlist from python?

DevBCN456, 2017-04-10:
I have realized in both cases volumio logs show don't work but in the setRandom to 0 by default restore the value and the web page shows the change by moving the icon color back to white from green.
I'm not much familiar with python, probably the problem is the way we have to pass the parameter. I see the API looks like this.. 
Does anyone knows how to pass the data parameter in data to get data.value to True?
			connWebSocket.on('setRandom', function (data) {
				//TODO add proper service handler
				var timeStart = Date.now();
				self.logStart('Client requests Volumio Random ' + data.value)
					.then(function () {
                        return self.commandRouter.volumioRandom(data.value);
					})
					.fail(self.pushError.bind(self))
					.done(function () {
						return self.logDone(timeStart);
					});
});

volumio, 2017-04-10:
setRandom, {value:true}

in any case, to debug use 

sudo journalctl -f

DevBCN456, 2017-04-10:
Fine this way it works...	Thx...
    command = 'setRandom'
    argument = {"value":"true"}
    socketIO.emit(command, argument)

closed 2017-04-10 by DevBCN456