volumio/Volumio2 #1129: [question] Turn off terminal on HDMI output?

codewarriorFX, 2017-04-10:
This may be a linux question, anyone able to turn off the terminal from the hdmi output? Im attempting to do this on Beaglebone and Rpi 3 versions.

xipmix, 2017-04-13:
What do you mean exactly? Why not just disconnect the hdmi cable?
To get rid of the login prompt you will have to disable the program behind it - agetty or some such.

codewarriorFX, 2017-04-13:
That's what I mean, to get rid of the login prompt. Not sure if is agetty...

xipmix, 2017-05-08:
This is a linux config issue more than anything, so probably this issue should be closed.

To get what you want is not trivial. The system really wants to have a console.

First turn off automatic starting of ttys
```
$ diff -u /etc/systemd/logind.conf.orig /etc/systemd/logind.conf
--- logind.conf.orig    2017-05-08 12:36:25.084608266 +0000
+++ logind.conf 2017-05-08 12:36:54.134528978 +0000
@@ -8,8 +8,8 @@
 # See logind.conf(5) for details
 
 [Login]
-#NAutoVTs=6
-#ReserveVT=6
+NAutoVTs=0
+ReserveVT=0
 #KillUserProcesses=no
 #KillOnlyUsers=
 #KillExcludeUsers=root
```
Then run ```sudo systemctl daemon-reload``` and reboot.

The other place that needs to change is the kernel boot line - it calls for a console on ```/dev/tty1``` and ```/dev/ttyAMA0``` (see ```/boot/cmdline.txt```).  Changing that will probably require running the volumio Build scripts yourself, which is not that hard. You may be able to work around by [disabling these ttys after boot](https://serverfault.com/questions/653886/how-to-disable-serial-console-after-successful-boot-systemd-grub2).

See also http://0pointer.de/blog/projects/serial-console.html