volumio/Volumio2 #1132: Search function

liberatorcnx, 2017-04-11:
The search function in Volumio always returns only ONE artist and only ONE album (assuming there are some matches to the search string).  Looking at the code it 'LOOKS' like someone has done a temporary fix to stop duplicates being queried which would crash, but it now exits the search at the first match (there is a 'return' statement which terminates the search for further matches).

I want to understand what has happened before with this because I have modified the mpd index file so that it works properly and returns ALL matching artists and ALL matching albums. I would appreciate some feedback before I submit my revision.

liberatorcnx, 2017-04-12:
I am away (in Thailand) right now so don't have all of my files etc.  Here is the revised index.js - maybe you can do a PR for it after you check it?  Please let me know if you have any qeries :-)  Otherwise it will need to wait for 2 weeks.
[Updated index.zip](https://github.com/volumio/Volumio2/files/915448/Updated.index.zip)
--
Steve (aka Liberator)

liberatorcnx, 2017-04-12:
OK - I managed to get the files I needed - PR done.

ev23, 2017-04-18:
I have the same issue when searching... ONE artist & ONE album. 
How did you solve it ?

liberatorcnx, 2017-04-18:
Hi ev23
I fixed a problem in mpd/index.js (see [PR 1134](https://github.com/volumio/Volumio2/pull/1134/files)).  I'm not sure if this has made it into the latest development release yet but assuming you are using the latest release you can simply replace /volumio/app/plugins/music_service/mpd/index.js with this file and it will work. Backup old file first!

There is one issue I noticed that still exists.  If you have duplicate album names then it will only show the first.  I have 3 albums called 'Greatest Hits' and only one shows, however clicking on that album shows tracks from all 3 albums.

I have fixed that and will do another PR soon that addresses this and a few other navigation problems.

closed 2017-06-02 by volumio