volumio/Volumio2 #1144: Fixing airplay problem with macos and web based videos

m125, 2017-04-17:
Hello,

I hade a problem with Youtube videos.
I looked one over Airplay. Syncing was fine.
Than i go on to the next one and every time the connection broke.
I've done some research and here we are.
I have an solution for this.
Please change: /volumio/app/plugins/music_service/airplay_emulation/shairport-sync.conf.tmpl 
old

```
general =
{
    name = "${name}";
log_verbosity = 0;
};

alsa =
{
  output_device = "${device}";
};

sessioncontrol =
{
  run_this_before_play_begins = "/usr/bin/mpc stop";
  allow_session_interruption = "yes";
};

```

new

```
general =
{
  name = "${name}";
  log_verbosity = 1;
  interpolation = "soxr";
};

alsa =
{
  output_device = "${device}";
};

sessioncontrol =
{
  run_this_before_play_begins = "/usr/bin/mpc stop";
  allow_session_interruption = "yes";
};

```

volumio, 2017-04-18:
Good, so the change is just to add:
interpolation = "soxr";

correct?

m125, 2017-04-18:
I think you also have to change
log_verbosity = 0 to 1
I tested it and it doesn't work without it.
I don't know why.

volumio, 2017-04-18:
Quite surprising indeed! Ok I'll add it 👍

closed 2017-04-18 by volumio

MarkusBec, 2017-05-24:
this is still not working only for like 40% of the time.